Keychron/qmk_firmware #287: [Feature Request] "Snap Tap"/"Rappy Snappy" for Lemokey L3 and other Keychron keyboards

Mraziiiik, 2024-07-23:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

"Snap Tap" is a functionality designed to enhance the gaming experience by improving directional key input handling. Here is a detailed explanation of how it works and its benefits:

With a standard gaming keyboard, when two opposing directional keys are pressed simultaneously (e.g., 'W' and 'S' or 'A' and 'D'), the character stops moving because no movement is registered. The character remains stationary until one of the keys is released.

Snap Tap Mode:

When "Snap Tap" mode is enabled, in any instance where two opposing directional keys are detected, the last input is registered as the priority. This means the character will change directions instantly without having to release the previous key, allowing for seamless and responsive movement. Super useful for strafing.

Given the QMK firmware's flexibility and the customization options this feature could be integrated as an additional mode within the firmware. 

Here i found two github projects: first implements the mode very simply where once you hold "A" and press "D", the "A" is unpressed. This however doesnt allow the player to hold "A" and spam "D" for super quick strafing because the A gets unpressed.  https://github.com/qmk/qmk_firmware/pull/24000

This allows the hold "A", spam "D" https://github.com/Xelus22/qmk_firmware/tree/key_interrupt_recover_playground

revunix, 2024-07-23:
Would love it if it got the Q1 HE.

cheapzzi, 2024-07-23:
Hi, I was just wondering if you would be able to give some tips how to get this functionality? I was trying to implement it on my Keychron V1, but I couldn't get it working.

adophoxia, 2024-07-23:
> Hi, I was just wondering if you would be able to give some tips how to get this functionality? I was trying to implement it on my Keychron V1, but I couldn't get it working.

This is already being discussed over in the QMK server on Discord. Suggest you go there to ask for help since that PR is related.

Psebcool, 2024-07-25:
Razer and Wooting just added this on all of theirs existing keyboards.

Nice feature !

PeterMortensen, 2024-07-25:
Related: *[\[Core\] Feature: Add key cancellation](https://github.com/qmk/qmk_firmware/pull/24000)*. #24000 (2024-06-26)

adophoxia, 2024-09-23:
> QMK server said they won't be working on this, the admin over there changed his mind, and refuses to merge the PR or give it anymore attention. 

The PR and the issue were locked because
1) People kept using them as troubleshooting threads when it deviated from their original purpose
2) The messages given in https://github.com/qmk/qmk_firmware/issues/24216 were in no way related to what the RFC was asking for. 
3) This project is based off of _**volunteer**_ work, meaning that the PR will be looked at when the team has time to do so. 

> He locks any threads related to snap tap. 

Unless the "threads" being referred to is this one here, https://github.com/qmk/qmk_firmware/issues/24204, this is because #24000 was already present, making that Issue redundant, hence this _**one and only**_ Issue being locked in the first place

Geczy, 2024-09-23:
I believe locking the discussion just because a few recent responses were off-topic wasn't the right decision. It felt like an overreaction by the admin. Regarding https://github.com/qmk/qmk_firmware/issues/24216

ZhunCn, 2024-11-08:
SOCD has been added to the keychron hall effect keyboards, found here: https://github.com/Keychron/qmk_firmware/commit/1a1cb9b70a6659332bce9c22f4f2e9c4bb81d16c

Wonder how difficult it is to port it to the other keyboards, as the current implementation uses the "HE mode" tab in Keychron Launcher to configure.