Keychron/qmk_firmware #288: [Bug] Q12 Max - Incorrect operation of power switch

ievsdoma, 2024-07-24:
### Describe the Bug

The keyboard is not operating in the expected way with some positions of power switch. The following happens:
1. Bluetooth selected - the keyboard operates in bluetooth mode as expected
2. Cable mode, cable not connected - keyboard is off as expected
3. Cable mode, cable connected - keyboard turns on and tries to connect via bluetooth. If it is unable to connect via bluetooth, it will not work. If it can connect via bluetooth, it will use bluetooth connection to send the keys to the computer rather than the cable. It still possible to switch the keyboard into boot mode and flash it in this state. 
4. Wireless mode - the keyboard turns on, but does not connect to the dongle. Holding Fn+4 does not do anything. If you connect the cable, it will use the cable connection to send the keys, but it seemingly operates in battery mode and switches off backlight quickly. 
I tried profiles 'default' and 'via'. As soon as I flash stock firmware back, everything works normally. 

### Keyboard Used

Q12 Max

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q12-max-qmk-via-wireless-custom-mechanical-keyboard

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: D:/dev/GitHub/qmk_firmware_keychron_2
Ψ Detected Windows 10 (10.0.22631).
Ψ QMK MSYS version: 1.7.2
Ψ Userspace enabled: False
Ψ Git branch: wireless_playground
Ψ Repo version: 0.14.29
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-09-18 10:01:17 +0000 --  (0e9d558)
Ψ - lib/chibios-contrib: 2022-10-03 18:09:41 +0200 --  (bb8356f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2022-09-19 18:02:44 +0200 --  (8d56ea3)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

DharMahn, 2024-08-18:
same with q6 max

PeterMortensen, 2024-12-28:
Re *"everything works normally"*: Also [mouse actions in 2.4 GHz mode](https://www.reddit.com/r/Keychron/comments/1f47p4b/qmk_mouse_keys_not_working_properly_while_in/)?

ievsdoma, 2025-01-26:
@PeterMortensen, thank you for your reply. I had updated the firmware in the wireless controller as you suggested. Firmware in dongle was up to date already. It did not help with the operation of the switch, but it did help with mouse horizontal scroll, so that's something. 
It feels like the problem is handling of the position of power/mode switch. In central position (cable mode) it works like in Bluetooth mode (left position). In right position (2.4G mode) it works like in cable mode.