Keychron/qmk_firmware #289: [Feature Request] Publish source code for K13 Pro ISO - Many keyboards missing ISO source

mvbentes, 2024-07-25:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hi,

The source code for K13 Pro ISO is missing, as is the source for many (most?) other ISO keyboards. For this particular keyboard I could find the source for the ANSI version in both the `bluetooth_playground` and the `wireless_playground` branches, but none for the ISO version. 

I need to be able to set TAPPING_TERM for mod tap and that cannot be accomplished via the launcher. Right now it is frequently activating unwantedly, making the function unusable.

I see this was already requested at #204, but left unanswered. I kindly request that you provide this source code, as QMK support was the main selling point for me.

Crimix, 2024-08-08:
I would like it as well, I was thinking to buy one, but since I cannot compile my own firmware customizations on top of it as I have been doing with my other Keychron boards, then I see no reason to even look at it again.

It is a shame, since they must have the configuration in-house to compile the firmware, so it seems a bit silly to not release it here

exstntlmsnthrp, 2024-08-09:
@Crimix Don't hesitate! It's a brilliant keyboard as is! I don't regret buying it! Although, when it first came out you could download the config and use that. Not sure if they still have it available or not. I do wish that it was upstreamed though, that would be a lot better for customizing.

Crimix, 2024-08-10:
@exstntlmsnthrp thank you, but I have some customizations that I use for work and without them it does not make much sense to buy it. I hope Keychron will see this and add the missing files for the ISO version

exstntlmsnthrp, 2024-08-19:
> > @exstntlmsnthrp thank you, but I have some customizations that I use for work and without them it does not make much sense to buy it. I hope Keychron will see this and add the missing files for the ISO version
> 
> A lot of the time even when they make it opensource, they don't give a way to compile it btw. *look at the new HE keyboard for example*

I think they're making it pretty clear that they didn't actually care about open source and using it more as a way to attract buyers versus provide anything to those of us who want to modify it.

Crimix, 2024-08-19:
> A lot of the time even when they make it opensource, they don't give a way to compile it btw. _look at the new HE keyboard for example_

I don't follow, from a quick peek at the new HE keyboard, the setup and folder structure looks exactly like for all the keyboards I have seen and got to compile.

My main issue is just that is should not be an extra effort for them to put up the keyboards in this repo, I would hope that they already have branches for the different keyboads they make firmware for, otherwise it seems like a process that could be _optimized_ quite a bit and _should be_

exstntlmsnthrp, 2024-08-19:
> My main issue is just that is should not be an extra effort for them to put up the keyboards in this repo, I would hope that they already have branches for the different keyboads they make firmware for, otherwise it seems like a process that could be _optimized_ quite a bit and _should be_

I didn't even bother to look at the HE. I already have 3 of their other keyboards. So I don't know anything about it except setting the release announcement. 

I completely agree with you. The whole process could be automated and even don't within GitHub, essentially producing minimal work for them. But they keep refusing to do so and don't communicate. So it leaves us guessing and assuming. Not to mention, frustrated.

Pokeylooted, 2024-08-20:
> > My main issue is just that is should not be an extra effort for them to put up the keyboards in this repo, I would hope that they already have branches for the different keyboads they make firmware for, otherwise it seems like a process that could be _optimized_ quite a bit and _should be_
> 
> I didn't even bother to look at the HE. I already have 3 of their other keyboards. So I don't know anything about it except seeing the release announcement.
> 
> I completely agree with you. The whole process could be automated and even don't within GitHub, essentially producing minimal work for them. But they keep refusing to do so and don't communicate. So it leaves us guessing and assuming. Not to mention, frustrated.

They didn't release 90% of their keyboards, I've been making the QMK dev team and also personally reaching out to them to release these, as they're legally obligated too. I mean I thought there would be more people who would stand with me, but mainly people even on r/mechanicalkeyboards just don't care enough.

Pokeylooted, 2024-08-20:
People on r/keychron did more, than people did on r/mechanicalkeyboards

PeterMortensen, 2024-08-22:
As noted, the source code for the two K13 Pro ISO variants has now been released.

closed 2024-12-22 by github-actions[bot]