Keychron/qmk_firmware #290: Macro delays not working over bluetooth

mvbentes, 2024-08-02:
### Describe the Bug

I have a macro configured that works just fine over cable connection, but gets corrupted over bluetooth.
I just made a simpler proof of concept and the bug holds. 
The simpler macro is as follows:
`{+KC_D}{-KC_D}{4000}{+KC_J}{-KC_J}`
Over cable it correctly outputs `dj` with proper timing, but over bluetooth it seems like the first key remains pressed for the duration of the delay, outputing:
`dddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddj'

### Keyboard Used

K13 Pro ISO RGB

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-k13-pro-qmk-via-wireless-custom-mechanical-keyboard-iso-layout-collection?variant=40558857322585

### Operating System

Windows 

### qmk doctor Output

```text
Not relevant. I am not compiling from qmk. I am using launcher.keychron.com
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I have already updated firmware to the latest version 1.0.1

I can not compile my own version of qmk firmware to test if that would solve the issue because source code for K13 Pro ISO was not released, as already requested on #289

lokher, 2024-08-20:
Please try this firmware
[keychron_k13_pro_iso_rgb_via.zip](https://github.com/user-attachments/files/16667817/keychron_k13_pro_iso_rgb_via.zip)

PeterMortensen, 2024-08-22:
Re *"I can not compile my own version of QMK firmware to test if that would solve the issue because source code for K13 Pro ISO was not released"*: Yeah, version 1.0.1 for K13 Pro ISO RGB has a release date of 2024-01-26.

The [2024-03-30 change](https://www.reddit.com/r/Keychron/comments/1emdp12/comment/lh6f1tl/) ([EBA583](https://github.com/Keychron/qmk_firmware/commit/eba583cfa71fd8ede7226253f81d0574443852f3)) positively fixes a problem that prevents Via macros from being used in Bluetooth mode.

The source code for the ISO variants has [now been released](https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/k13_pro/iso/rgb) (2024-08-19).

mvbentes, 2024-09-19:
I just tested the firmware @lokher provided and it solved the macro timing issue. Thank you very much!
To anyone also interested, the firmware provided here is not the same as the one launcher.keychron offers as upgrade. That one was last updated july 2024.

I have not had the opportunity to test compiling from the now published source. I hope to have the time to do it, since I also wanted to adjust the timing of the mod tap function.

Best regards!

mvbentes, 2024-09-23:
Hi, @lokher!

I am having a different problem with this firmware you sent. I am having a LOT more unwanted activation on my mod-tap key, to the point where I am considering deactivating it. 

Was there any change to the timing limit in this version?