Keychron/qmk_firmware #293: Inconsistencies in code-base for some Lemokey X boards

adophoxia, 2024-08-13:
### Issue Description

Looking at the code for the X5 and X6, whatever they have doesn't match with their physical attributes.

X5:
- A 96% keyboard, but most of the code in 
  - [`info.json`](https://github.com/Keychron/qmk_firmware/blob/playground/keyboards/keychron/x5/info.json#L27-L119)/[`keyboard.json`](https://github.com/Keychron/qmk_firmware/blob/playground/keyboards/keychron/x5/ansi/red/info.json)
  - [`config.h`](https://github.com/Keychron/qmk_firmware/blob/playground/keyboards/keychron/x5/ansi/red/config.h#L25)

reference it to be as if it's a 75% keyboard when in reality, it is not. The same can be applied for its [`readme.md`](https://github.com/Keychron/qmk_firmware/blob/playground/keyboards/keychron/x5/readme.md?plain=1#L5).

X6:
- A 65% keyboard, but the code in
  - [`info.json`](https://github.com/Keychron/qmk_firmware/blob/playground/keyboards/keychron/x6/info.json#L28-L139)/[`keyboard.json`](https://github.com/Keychron/qmk_firmware/blob/playground/keyboards/keychron/x6/ansi/red/info.json#L28-L135)
  - [`config.h`](https://github.com/Keychron/qmk_firmware/blob/playground/keyboards/keychron/x6/ansi/red/config.h#L25)

reference it to be as if it's a 96% keyboard when in reality, it is not. The same can be applied for its [`readme.md`](https://github.com/Keychron/qmk_firmware/blob/playground/keyboards/keychron/x6/readme.md?plain=1#L5).

Was this done intentionally with custom code that isn't visible in the branch, or was this an oversight and done so without vetting what keyboard is exactly what? I would love to get an answer to this so as to not bring any confusion.

eleniums, 2024-11-25:
I own a Lemokey X6 and encountered this same issue where the code and design files appear to be incorrect. Can we get the corrected files committed?

osu761, 2025-01-21:
2025 and these still have not been corrected

eleniums, 2025-01-24:
FYI, the Lemokey X6's correct design file was finally uploaded to their firmware page:
https://www.lemokey.com/pages/firmware-and-json-files-of-the-lemokey-qmk-keyboards

Here's the direct link to the X6 design file:
https://cdn.shopify.com/s/files/1/0059/0630/1017/files/x6_ansi_red_v1.0.0_2402211427.json.zip?v=1733219270

With this new design file, I am finally able to use Via to configure my X6. I would still like to see the correct firmware source code made available somewhere.