Keychron/qmk_firmware #304: [Bug] Q15 Max battery level animation only going up to 50%

MinaDarsh, 2024-09-01:
### Describe the Bug

When building firmware using any of the keymaps for the Q15 Max from the [abefc1d commit](url=https://github.com/Keychron/qmk_firmware/commit/abefc1d4a6be3ca70fec42288d5f30896ba70a81) and flashing it to the board, the battery level animation will only go up to 5 lights (50%) even if the charging LED on the back has indicated a full charge (green).

### Keyboard Used

q15_max/ansi_encoder/keymaps/*

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q15-max-qmk-wireless-custom-mechanical-keyboard

### Operating System

Windows 11

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: C:/Users/Mina/Keychron/qmk_firmware
Ψ Detected Windows 10 (10.0.22621).
Ψ QMK MSYS version: 1.7.2
Ψ Userspace enabled: False
Ψ Git branch: wireless_playground
Ψ Repo version: 0.14.29
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-01-10 10:36:30 +0800 --  (86a3ed9ba)
Ψ - lib/chibios-contrib: 2024-06-04 11:39:47 +0800 --  (5dbc2a07)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

lokher, 2024-09-02:
Fixed, please fetch and pull.

MinaDarsh, 2024-09-02:
That did the trick, thank you!

closed 2024-09-02 by MinaDarsh