Keychron/qmk_firmware #310: [Bug] Keychron Q3 HE LED Config Incorrect

KellyNyanbinary, 2024-10-09:
### Describe the Bug

The LED Config for the 80% Q3 HE uses the layout for Q1 HE, a 75% keyboard.

### Keyboard Used

keychron/q3_he/ansi_encoder

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q3-he-qmk-wireless-custom-keyboard

### Operating System

Ububtu 24.04

### qmk doctor Output

```text
/home/kelly/.local/Lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
milc.set_metadata(version =_ version_)
QMK Doctor is checking your environment.
CLI version: 1.1.5
QMK home: /home/kelly/qmk_firmware
Detected Linux (WSL, Ubuntu 24.04.1 LTS).
Userspace enabled: False
Git branch: hall_effect_playground
Repo version: 0.14.29
A Git has unstashed/uncommitted changes.
A The official repository does not seem to be configured as git remote "upstream".
All dependencies are installed.
Found arm-none-eabi-gcc version 13.2.1
Found avr-gcc version 7.3.0
Found avrdude version 7.1
Found dfu-programmer version 0.6.1
Found dfu-util version 0.11
Would you like to clone the submodules? [Y/n] Y
Submodule status:
- lib/chibios: 2024-01-10 10:36:30 +0800 -- (86a3ed9ba)
- Lib/chibios-contrib: 2023-07-17 11:39:05 +0200 -- (da78eb37)
- lib/googletest: 2021-06-11 06:37:43 -0700 -- (e2239ee6)
- lib/lufa: 2022-08-26 12:09:55 +1000 -- (549b97320)
- lib/vusb: 2022-06-13 09:18:17 +1000 -- (819dbc1)
- lib/printf: 2022-06-29 23:59:58 +0300 -- (c2e3b4e)
- lib/pico-sdk: 2023-02-12 20:19:37 +0100 -- (a3398d8)
- lib/lvgl: 2022-04-11 04:44:53 -0600 -- (e19410f8)
QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

## Detail

What is currently there:
```C
led_config_t g_led_config = {
    {
        // Key Matrix to LED Index
        {  0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15 },
        { 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31 },
        { 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47 },
        { 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, __, 60, 61, 62 },
        { 63, __, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, __, 74, __, 75 },
        { 76, 77, 78, __, __, __, 79, __, __, 80, 81, 82, 83, 84, 85, 86 },
    },
    {
        // LED Index to Physical Position
        {0, 0}, {16, 0}, {29, 0}, {42, 0}, {55, 0}, {71, 0}, {84, 0}, { 97, 0}, {110, 0}, {126, 0}, {139, 0}, {152, 0}, {165, 0},           {198, 0}, {211, 0}, {224, 0},
        {0,15}, {13,15}, {26,15}, {39,15}, {52,15}, {65,15}, {78,15}, { 91,15}, {104,15}, {117,15}, {130,15}, {143,15}, {156,15}, {176,15}, {198,15}, {211,15},
        {3,28}, {19,28}, {33,28}, {46,28}, {59,27}, {72,27}, {85,27}, { 98,28}, {111,28}, {124,28}, {137,28}, {150,28}, {163,28}, {179,28}, {198,28}, {211,27},
        {5,40}, {23,40}, {36,40}, {49,40}, {62,40}, {75,40}, {88,40}, {101,40}, {114,40}, {127,40}, {140,40}, {153,40},           {174,40}, {224,15}, {224,27},
        {3,52},          {29,52}, {42,52}, {55,52}, {68,52}, {81,52}, { 94,52}, {107,52}, {120,52}, {133,52}, {146,52},           {179,52},           {211,52},
        {2,64}, {18,64}, {34,64},                            {83,64},                               {131,64}, {148,64}, {164,64}, {180,64}, {198,64}, {211,64}, {224,64}
    },
    {
        // RGB LED Index to Flag
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,    1, 1, 1,
        1,    1, 1, 1, 1, 1, 1, 1, 1, 1, 1,    1,    1,
        1, 1, 1,          1,       1, 1, 1, 1, 1, 1, 1
    }
};
```

What should probably be there (from Q3 Max):
```C
led_config_t g_led_config = {
    {
        // Key Matrix to LED Index
        {  0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, __, 13, 14, 15 },
        { 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32 },
        { 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49 },
        { 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, __, 62, __, __, __ },
        { 63, __, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, __, 74, __, 75, __ },
        { 76, 77, 78, __, __, __, 79, __, __, __, 80, 81, 82, 83, 84, 85, 86 },
    },
    {
        // LED Index to Physical Position
        {0, 0}, {16, 0}, {29, 0}, {42, 0}, {55, 0}, {71, 0}, {84, 0}, { 97, 0}, {110, 0}, {126, 0}, {139, 0}, {152, 0}, {165, 0},           {198, 0}, {211, 0}, {224, 0},
        {0,15}, {13,15}, {26,15}, {39,15}, {52,15}, {65,15}, {78,15}, { 91,15}, {104,15}, {117,15}, {130,15}, {143,15}, {156,15}, {176,15}, {198,15}, {211,15}, {224,15},
        {3,28}, {19,28}, {32,28}, {46,28}, {59,28}, {72,28}, {85,28}, { 98,28}, {111,28}, {124,28}, {137,28}, {150,28}, {163,28}, {179,28}, {198,28}, {211,28}, {224,28},
        {5,40}, {23,40}, {36,40}, {49,40}, {62,40}, {75,40}, {88,40}, {101,40}, {114,40}, {127,40}, {140,40}, {153,40},           {174,40},
        {8,52},          {29,52}, {42,52}, {55,52}, {68,52}, {81,52}, { 94,52}, {107,52}, {120,52}, {133,52}, {146,52},           {171,52},           {211,52},
        {2,64}, {18,64}, {34,64},                            {83,64},                               {131,64}, {148,64}, {164,64}, {180,64}, {198,64}, {211,64}, {224,64},
    },
    {
        // RGB LED Index to Flag
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,    1, 1, 1,
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,    1,
        1,    1, 1, 1, 1, 1, 1, 1, 1, 1, 1,    1,    1,
        1, 1, 1,          1,          1, 1, 1, 1, 1, 1, 1,
    }
};
```

## What I Tried

I tried fixing the LED config by copying and pasting the code from Q3 Max's config, but QMK is complaining about a mismatched LED matrix size.

lokher, 2024-11-08:
Did you mean the last part "RGB LED Index to Flag"?
I tried the code from q3_max, no error.

We mapped the flag position to first part.
[image]

KellyNyanbinary, 2024-11-16:
The `// Key Matrix to LED Index` array and the two subsequent arrays do not match the actual LED layout of the Q3 HE.

lishan89uc, 2025-04-26:
did this ever get resolved? I am looking into adding per key rgb to the q3 he