Keychron/qmk_firmware #314: [Bug] Macro delays not working with K6 Pro ISO

farconada, 2024-10-11:
### Describe the Bug

[image]
Macro delays dont work with the K6 Pro ISO version

### Keyboard Used

Keychron K6 Pro ISO ES

### Operating System

Linux

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

farconada, 2024-10-24:
please, help

PeterMortensen, 2024-11-10:
Re *"not working"*: Can you be more specific?

In what way are macro delays not working? For example, 

 * Are macro delays not accepted by the [Via clone](https://www.reddit.com/r/Keychron/comments/1buwibx/comment/kxx3kt0/)? Is there the same problem in [Via](https://docs.keeb.io/via)?
 * Are the macro delays not recorded during macro recording?
 * Do the macro delays appear to be entered, but don't have any effect when the macro is executed?
 * What is the macro source for the macro? Example (delays 17 ms and 400 ms):
     * `{+KC_F24}{17}{-KC_F24}{400}c`

farconada, 2024-11-10:
[image]
With keychron launcher, but when I reconnect the delay has dissapeared. I have pushed "submit"
[image]

PeterMortensen, 2024-11-11:
Re *"...when I reconnect the delay has disappeared. I have pushed "submit""*: It is mostly likely a problem with the Via clone. It could be confirmed by doing the same with Via (that would exclude the keyboard as the cause).

PeterMortensen, 2025-10-05:
I recommend compiling the keyboard firmware from source code, later [than 2024-03-30](https://www.reddit.com/r/Keychron/comments/19fn0j3/comment/lc3lh1p/).

Related:

 * *[\[Bug\] Compile error "eeconfig_language.h: No such file or directory" #396](https://github.com/Keychron/qmk_firmware/issues/396)*

A workaround is to use branch ***"[wls_2025q1](https://github.com/Keychron/qmk_firmware/tree/wls_2025q1/keyboards/keychron/q5_pro)"*** instead. Note [the branch confusion](https://www.reddit.com/r/Keychron/comments/1j092s2/comment/mvu2xyk/).