Keychron/qmk_firmware #317: Changelog

fulldecent, 2024-10-24:
### Issue Description

Please do releases and publish a changelog for this repo

This page explains the upgrade process which is a pain https://www.keychron.com/pages/firmware-and-json-files-of-the-keychron-qmk-v-and-v-max-series-keyboards (Macs don't install unsigned software!)

and Keychron V1 does not work out-of-the box on Via https://github.com/the-via/app/issues/91

So I'd it would be nice to explain what features you get for all this pain

PeterMortensen, 2024-11-15:
Re *"Keychron V1 does not work out-of-the box on Via"*: Are you sure?

The whole Keychron V series is supposed to work out of the box with Via.

I have never had any problems with a V6 (that I am typing this on). That is on [Linux](https://en.wikipedia.org/wiki/Ubuntu_%28operating_system%29), with [the standalone Via application](https://pmortensen.eu/world2/2023/06/12/via-a-mechanical-qmk-keyboard-configuration-works-just-fine-without-google-chrome/) and with the QMK development environment installed (which supposedly sets up the [udev](https://en.wikipedia.org/wiki/Udev) rules, etc.).

fulldecent, 2024-11-15:
The issue is open there on the Via repo and linked above, several people seem to have the problem. Perhaps it is older inventory of the V's that have old firmware? But I'm not sure.

closed 2025-03-16 by github-actions[bot]