Keychron/qmk_firmware #322: [Bug] cant create default Keymap

ChoosenMEME, 2024-11-06:
### Describe the Bug

everytime i run `make keychron/q1_he/ansi_encoder:default`

i get the following output
```
QMK Firmware 0.14.29
Making keychron/q1_he/ansi_encoder with keymap default

arm-none-eabi-gcc.exe (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: platforms/chibios/drivers/i2c_master.c                                                   [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/eeprom_he.c                                    [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/analog_matrix_scan.c                           [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/profile.c                                      [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/usb_descriptor_override.c                      [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/action_regular_trigger.c                       [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/action_rapid_trigger.c                         [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/action_okmc.c                                  [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/action_toggle.c                                [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/action_joystick.c                              [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/action_xinput.c                                [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/action_socd.c                                  [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/sqrt.c                                         [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/game_controller_common.c                       [OK]
Compiling: ./keyboards/keychron/common/analog_matrix/analog_matrix.c                                [OK]
Compiling: ./keyboards/keychron/common/keychron_task.c                                              [OK]
Compiling: ./keyboards/keychron/common/keychron_common.c                                            [OK]
Compiling: ./keyboards/keychron/common/keychron_raw_hid.c                                          ./keyboards/keychron/common/keychron_raw_hid.c: In function 'kc_raw_hid_rx':
./keyboards/keychron/common/keychron_raw_hid.c:88:20: error: 'id_get_keyboard_value' undeclared (first use in this function)
   88 |     if (data[0] == id_get_keyboard_value && data[1] == id_switch_matrix_state) {
      |                    ^~~~~~~~~~~~~~~~~~~~~
./keyboards/keychron/common/keychron_raw_hid.c:88:20: note: each undeclared identifier is reported only once for each function it appears in
./keyboards/keychron/common/keychron_raw_hid.c:88:56: error: 'id_switch_matrix_state' undeclared (first use in this function)
   88 |     if (data[0] == id_get_keyboard_value && data[1] == id_switch_matrix_state) {
      |                                                        ^~~~~~~~~~~~~~~~~~~~~~
 [ERRORS]
 |
 |
 |
make[1]: *** [builddefs/common_rules.mk:373: .build/obj_keychron_q1_he_ansi_encoder_default/common/keychron_raw_hid.o] Error 1
Make finished with errors
make: *** [Makefile:416: keychron/q1_he/ansi_encoder:default] Error 1
```

### Keyboard Used

keychron/q1_he

### Operating System

Windows 11

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: D:/Tools/qmk_firmware
Ψ Detected Windows 10 (10.0.22631).
Ψ QMK MSYS version: 1.9.0
Ψ Userspace enabled: False
Ψ Git branch: hall_effect_playground
Ψ Repo version: 0.14.29
Ψ - Latest hall_effect_playground: 2024-11-06 13:18:56 +0800 (9cc6233447) -- Merge pull request #313 from AnalogSense/fix-analogue-sticks
Ψ - Latest upstream/master: None
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: None
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-01-10 10:36:30 +0800 --  (86a3ed9ba)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb37)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

ChoosenMEME, 2024-11-06:
removing lines 88 to 91 from [qmk_firmwareke/yboards/keychron/common/keychron_raw_hid.c](https://github.com/Keychron/qmk_firmware/blob/hall_effect_playground/keyboards/keychron/common/keychron_raw_hid.c)
```
    if (data[0] == id_get_keyboard_value && data[1] == id_switch_matrix_state) {
        send_analog_matrix(data, length);
        return false;
    }
```
removed the build error

lokher, 2024-11-08:
change line 87 to `#if defined(ANANLOG_MATRIX) && defined(VIA_ENABLE)`.
the fix has been pushed.

closed 2024-11-08 by ChoosenMEME