Keychron/qmk_firmware #323: [Feature Request] K8 Max Source Code

venengaandrew, 2024-11-08:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

any chance we could get the source code for the K8 Max?

lokher, 2024-11-08:
Added.
https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/k8_max

venengaandrew, 2024-11-08:
Thank you!

Quick issue, flashing the RGB ANSI profile to my K8 Max makes keys 1, Q, and A unresponsive. 

Is there any debug information I could provide for more clarity?

This was tested running both: 
make keychron/k8_max/ansi/rgb:default and flashing using QMK Toolbox
make keychron/k8_max/ansi/rgb:default:flash

lokher, 2024-11-09:
> Thank you!
>
> Quick issue, flashing the RGB ANSI profile to my K8 Max makes keys 1, Q, and A unresponsive.
>
> Is there any debug information I could provide for more clarity?
>
> This was tested running [...]

Thanks for your feedfback, it's fixed,  please pull and compile again.

closed 2024-11-09 by venengaandrew