Keychron/qmk_firmware #326: [Feature Request] How to support key remapping for Keychron K12J

litao3rd, 2024-11-12:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hi, there. I have a K12J Keychron keyboard with 61 keys, and I really like the Poker layout. I've become accustomed to using FN1 + WASD as arrow keys, but this isn't configured that way on the K12J. I also need the programmable key support that the Poker layout offers. I would like to ask if the K12J supports key remapping and programmable functions, or if there’s any way I can achieve this. I even considered designing a PCB of the same size to replace the K12J's PCB and writing custom firmware, but that seems like a huge amount of work.

PeterMortensen, 2024-11-13:
Re *"if the K12J supports key remapping and programmable functions, or if there’s any way I can achieve this."*: No, not in a standard way.

Some options:
 * Go the [exotic route](https://www.reddit.com/r/olkb/comments/1avk4g3/comment/krigncf/). It may not be on [the list of supported keyboards](https://jpe230.github.io/SonixQMKDocs/compatible_kb/#keychron), but perhaps they are just waiting for someone like you with the particular hardware in order to add support, test it, etc. Wireless connectivity is not an option at the moment, but they are allegedly working on it.
 * Repurpose it with another microcontroller and QMK (or similar, like [ZMK](https://github.com/zmkfirmware/zmk) or [FAK](https://github.com/semickolon/fak)). The existing circuitry needs to be sufficiently inactivated for this to work. This should at least get the basic functionality of a fully programmable keyboard. Wireless connectivity and RGB light may be more difficult. The most realistic is probably ZMK and a suitable microcontroller, like [Nice!Nano](https://splitkb.com/products/nice-nano).

Re *"designing a PCB of the same size to replace the K12J's PCB and writing custom firmware"*:

PeterMortensen, 2024-11-13:
Though it is probably not the right place. This issue should probably be closed.

litao3rd, 2024-11-14:
Hi, thanks very much for your guidelines. I will try to repurpose the keyboard with another microcontroller like you said.

closed 2024-11-14 by litao3rd