Keychron/qmk_firmware #327: Keycodes for Space Cadet Keys seems wrong (Keychron Q3 HE)

drandarov-io, 2024-11-14:
### Issue Description

https://docs.qmk.fm/keycodes describes the space cadet keycodes as starting with `SC_` , however inside https://launcher.keychron.com/#/keymap the keycodes they start with `KC_`.

This caused me quite a bit of confusion until I went over every single key in the launcher since the codes from the docs didn't work.
Not sure if this is a general issue or just with the Q3 HE.

[image]

[image]

PeterMortensen, 2024-11-15:
The [Via clone](https://www.reddit.com/r/Keychron/comments/1buwibx/comment/kxx3kt0/) is, well, a clone of Via, so what did you expect? Yes, that is a rhetorical question (mostly).

That is in ***KEYMAP*** &rarr; ***SPECIAL*** &rarr; "LA(" (about 15% down in the list).

PeterMortensen, 2024-11-15:
An initial guess would be that some renaming happened in QMK, e.g., like for some keycodes for mouse actions ([2024-07-03](https://github.com/qmk/qmk_firmware/pull/23975)).

lokher, 2024-11-29:
It's fixed now, thank you for pointing it out.

drandarov-io, 2024-12-05:
[image]

While we're at it: This seems to be a more widespread problem. In this case `QK_GRAVE_ESCAPE`, `QK_GESC`, as well as `KC_GRAVE_ESCAPE`. Do not work. 

Only the alias with the `QK` changed to `KC` works, meaning: `KC_GESC`, but notable not `KC_GRAVE_ESCAPE`.

closed 2025-04-05 by github-actions[bot]