Keychron/qmk_firmware #329: c2_pro help needed with indicator lights

Vasja1986, 2024-11-17:
### Issue Description

I have a c2 pro rgb keyboard. When compiling the default keymap for this keyboard and subsequently flashing the compiled code into the keyboard, the indicator lights (num, caps, mac, win) do not show the right indications, instead they are flashing with a pattern that follows the rgb matrix. How can i get the indicator lights behave like in the default factory firmware, i.e. caps only lights when caps lock is activated etc?

PeterMortensen, 2024-11-22:
Does it work when the RGB animation mode is static? ("[Solid colour](https://pmortensen.eu/world2/2023/06/19/keychron-v5-a-reasonably-priced-fully-macro-capable-qmk-based-mechanical-keyboard#LightingModes)")

PeterMortensen, 2024-11-22:
Re *"they are flashing with a pattern that follows the RGB matrix"*: Can you elaborate? What happens when doing specific actions? What actions? Can you provide an example?

Vasja1986, 2024-11-22:
> Re _"they are flashing with a pattern that follows the RGB matrix"_: Can you elaborate? What happens when doing specific actions? What actions? Can you provide an example? In what RGB animation mode?
It does not respond to any actions. The default animation (whatever that is) gets displayed if RGB_MATRIX is enabled and does not react to input.

Vasja1986, 2024-11-27:
Turns out the problem is in qmk_firmware/tree/master/keyboards/keychron/c2_pro/c2_pro.c

the indicator code is commented out if LED_MATRIX_ENABLE is not defined, which is the case for the rgb variant of the keyboard. The last #endif should be moved up some place before `bool process_record_kb`. 

[rant]Great waste of time. Excellent quality control.[/rant]

closed 2025-03-28 by github-actions[bot]