Keychron/qmk_firmware #330: [Bug] Not able to build make keychron/k2_pro/ansi/rgb:default keyboard

patelgunjan007, 2024-11-20:
### Describe the Bug

Hello Team,
I am using linux 22.04 LTS system
I have cloned the master branch and tried to build code using below command but its not working.

make keychron/k2_pro/ansi/rgb:default;

and k2_pro is not showing into the keyboard list.

Ad I am not able to launch my keyboard using luncher(https://launcher.keychron.com/#/keymap) aslo.

I wan to chech my keymap for my k2_pro(https://keychron.in/product/keychron-k2-pro-white-qmk-via-wireless-mechanical-keyboard/)

Could you please guide and fix the bug if required.

### Keyboard Used

make keychron/k2_pro/ansi/rgb:default;

### Link to product page (if applicable)

https://keychron.in/product/keychron-k2-pro-white-qmk-via-wireless-mechanical-keyboard/

### Operating System

ubuntu 22.04 LTS

### qmk doctor Output

```text
❯ git checkout master
Updating files: 100% (39026/39026), done.
Previous HEAD position was 9472c4b8cb Fix US_RDQU keycode (#13464)
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
❯ make keychron/k2_pro/ansi/rgb:default
QMK Firmware 0.14.29
/home/gunjan/.local/lib/python3.8/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/gunjan/.local/lib/python3.8/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
make: *** No rule to make target 'keychron/k2_pro/ansi/rgb:default'. Stop.
|
| QMK's make format is:
|     make keyboard_folder:keymap_folder[:target]
|
| Where `keyboard_folder` is the path to the keyboard relative to
| `qmk_firmware/keyboards/`, and `keymap_folder` is the name of the
| keymap folder under that board's `keymaps/` directory.
|
| Examples:
|     keyboards/dz60, keyboards/dz60/keymaps/default
|       -> make dz60:default
|       -> qmk compile -kb dz60 -km default
|     keyboards/planck/rev6, keyboards/planck/keymaps/default
|       -> make planck/rev6:default:flash
|       -> qmk flash -kb planck/rev6 -km default
|
/home/gunjan/.local/lib/python3.8/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

patelgunjan007, 2024-11-20:
I found the solution
 thanks

closed 2024-11-20 by patelgunjan007

PeterMortensen, 2024-11-22:
> I found the solution thanks

Can you share the solution with us, please? So others can learn from it.

petermao, 2024-12-23:
I saw the exact same linking errors when trying to compile for a keychron v6_max.

The solution was to run `qmk git-submodule` after `qmk setup`.  Sadly, I don't know how I came across this as a solution.  I've already forgotten once and had to dig it out of my `~/.bash_history`!