Keychron/qmk_firmware #333: Publish source code for Q10 Max ISO

smazh, 2024-11-27:
### Issue Description

I'd like to request source files for the [ISO variant](https://www.keychron.uk/products/keychron-q10-max-alice-layout-qmk-custom-mechanical-keyboard-iso-layout-collection) of the Q10 Max please. The ANSI version has sources available at https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/q10_max but the iso_encoder appears to be missing.

lokher, 2024-11-29:
Added
https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/k10_max/iso

smazh, 2024-11-29:
Thank you! For everyone's benefit correct link for Q10 max iso: https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/q10_max/iso_encoder

closed 2024-11-29 by smazh

lokher, 2024-11-29:
link updated too.