Keychron/qmk_firmware #335: [Bug] V4 Max double typing on mod-tap and layer-tap only on wireless

dclin02, 2024-11-28:
### Describe the Bug

When I configure any key as a mod-tap or layer-tap, if the keyboard is in wireless mode (both wireless dongle or bluetooth), I get double/duplicated key presses after pressing the mod/layer-tap key, for a small window, also duplicating the next key as well if it is pressed quickly enough.

Other keys keep working correctly without duplication - it only occurs upon pressing a mod/layer-tap. If I change to a layer that doesn't have a layer/mod-tap, it never happens.

Below is a gif illustrating the problem - for this gif I configured `space` as `LT(4, KC_SPC)`, and the rest are default.
[image]

Every `space` physical key press ends up as two spaces, with at least the next key also duplicated.

### Keyboard Used

V4 Max

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-v4-max-qmk-via-wireless-custom-mechanical-keyboard

### Operating System

Windows and Linux

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I got a new V4 max recently, and have not flashed modified firmware to it, only having used the `launcher.keychron.com` website.

At first, everything works as expected. When I configure any key as a mod-tap or layer-tap, as long as I'm connect through cable, everything keeps working correctly, but as soon as I change to the wireless dongle, or to bluetooth, I get the described bug.

I tested it both on Linux and Windows and it happened on both.

I'm on firmware 1.0.0 and `wireless firmware` 0.2.1.

lokher, 2024-11-29:
Can you try the firmware on github?
https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron/v4_max/firmware

dclin02, 2024-11-29:
Just to confirm, I can just flash the bin in https://github.com/Keychron/qmk_firmware/blob/wireless_playground/keyboards/keychron/v4_max/firmware/keychron_v4_max_ansi_via.bin and VIA/launcher.keychron should keep working?

lokher, 2024-11-30:
> Just to confirm, I can just flash the bin in https://github.com/Keychron/qmk_firmware/blob/wireless_playground/keyboards/keychron/v4_max/firmware/keychron_v4_max_ansi_via.bin and VIA/launcher.keychron should keep working?

Sure thing.

dclin02, 2024-12-01:
Flashing the firmware on GitHub has resolved the issue. Thank you.

closed 2024-12-01 by dclin02