Keychron/qmk_firmware #338: [Feature Request] BLE Advertising Battery Level

spit4520, 2024-12-03:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

Hello,

I have the Keychron K10 Pro, and I love it. I would love it if I could submit a PR with the necessary changes to expose the the proper GATT service for the Battery Level. I have been writing automotive, infotainment, and embedded for for 10 years and I am willing to donate the feature. If you could please let me know if this is the correct repository so submit the PR, and give me the specifications you would require in order to merge this upstream, then I can get started. I understand that battery life could be a concern, so if you could provide your internal product goals on the refresh interval for the advertisement then I can make sure the PR is compliant.

I've loved my Keychron and this is the only feature that I have found in the last year that has been missing, you are doing a wonderful job and I'd love to volunteer to keep the ball rolling

Best,

S.Joseph

PeterMortensen, 2024-12-06:
Presuming it is the same thing, [it works for the V6 Max](https://pmortensen.eu/world2/2024/11/04/the-battery-state-of-a-keychron-qmk-based-keyboard-can-be-displayed-in-the-operating-system/) (presumably the whole V Max series. And probably also the K Max and Q Max series).

[Proof](https://pmortensen.eu/world2/wp-content/uploads/2024/11/Keychron_V6_Max_battery_84_percent_Ubuntu_panel_Settings_Power.png).

But [not for the K Pro series](https://pmortensen.eu/world2/2024/11/04/the-battery-state-of-a-keychron-qmk-based-keyboard-can-be-displayed-in-the-operating-system/#NotForKeychronProSeries) in the same setup.

Presumably, they use different Bluetooth modules. I don't know if this has anything to do with it.

bam80, 2024-12-13:
> But [not for the K Pro series](https://pmortensen.eu/world2/2024/11/04/the-battery-state-of-a-keychron-qmk-based-keyboard-can-be-displayed-in-the-operating-system/#NotForKeychronProSeries) in the same setup.

Might be related, please confirm:
#341

PeterMortensen, 2024-12-27:
> > But [not for the K Pro series](https://pmortensen.eu/world2/2024/11/04/the-battery-state-of-a-keychron-qmk-based-keyboard-can-be-displayed-in-the-operating-system/#NotForKeychronProSeries) in the same setup.
> 
> Might be related, please confirm: #341

For a [K5 Pro](https://www.keychron.com/products/keychron-k5-pro-qmk-via-wireless-custom-mechanical-keyboard-iso-layout-collection) and [Keychron V6 Max](https://www.keychron.com/products/keychron-v6-max-qmk-via-wireless-custom-mechanical-keyboard-iso-layout-collection) connected at the same time, I get, respectively ("EE:0A:F9:0F:92:FC" is the V6 Max),

    bluetoothd -v
    sudo btmgmt con 

Output:
    5.53
    6C:93:08:65:8E:E6 type BR/EDR
    EE:0A:F9:0F:92:FC type LE Random

That is for a Bluetooth 5.3 adapter ([PCIe](https://en.wikipedia.org/wiki/PCI_Express), but the Bluetooth part is internally connected through USB).

bam80, 2024-12-27:
> 6C:93:08:65:8E:E6 type BR/EDR

So this is the problem - BR/EDR corresponds to Bluetooth 2.0/2.1, no BLE connection in this mode, and no battery services available.

https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/Assigned_Numbers/out/en/Assigned_Numbers.pdf?v=1733417984269
https://www.mathworks.com/help/bluetooth/gs/comparison-of-bluetooth-bredr-and-bluetooth-le.html

PeterMortensen, 2025-01-30:
Perhaps they copy-pasted [the claim](https://www.keychron.com/products/keychron-k5-pro-qmk-via-wireless-custom-mechanical-keyboard-iso-layout-collection) for the K Pro series from the K series, [which seem to have it](https://github.com/IulianDita/keychron/blob/master/k2.md#battery)?

With the V6 Max connected (and [working properly](https://www.reddit.com/r/Keychron/comments/1g1698j/comment/m14b9bc/)):

    upower --dump | grep keyboard -A 7

Output:

    Device: /org/freedesktop/UPower/devices/keyboard_dev_DD_4B_7B_83_76_EE
      native-path:          /org/bluez/hci0/dev_DD_4B_7B_83_76_EE
      model:                Keychron V6 Max
      serial:               DD:4B:7B:83:76:EE
      power supply:         no
      updated:              Thu 01 Jan 1970 01:00:00 CET (1738250122 seconds ago)
      has history:          yes
      has statistics:       no
      keyboard
        present:             yes
        rechargeable:        no
        state:               unknown
        warning-level:       none
        percentage:          90%
        icon-name:          'battery-missing-symbolic'

With the K5 Pro connected (and working properly), the output is empty.

PeterMortensen, 2025-02-27:
From [another comment](https://github.com/Keychron/qmk_firmware/issues/341#issuecomment-2646135823):

> "the Bluetooth module of the K series ckbt51 is not compatible with BLE"

closed 2025-06-29 by github-actions[bot]