Keychron/qmk_firmware #344: ZMK Source code request: Keychron B1 Pro Ultra-Slim Wireless Keyboard

bam80, 2024-12-18:
This product claims "ZMK support":
https://www.keychron.com/products/keychron-b1-pro-ultra-slim-wireless-keyboard

I see this ZMK fork but there is no Keychron code in it:
https://github.com/Keychron/zmk

Could you please publish the sources?

bam80, 2025-01-26:
@4cio That's great, thanks for share!

Please make corresponding issues there if you unable to build, it's quite important. Thanks.

bam80, 2025-01-26:
After someone would be able to build and flash working firmware with it, I think we could close this issue.

bam80, 2025-01-26:
> To build and flash the the firmware, hold down the RESET button while connecting the keyboard over USB. The run:
> ```sh
> west build -p -b keychron && west flash
> ```

https://github.com/Keychron/zmk/blob/dd86028ef213db5d9d3feabe10ae6ac256040d8a/app/boards/arm/keychron/README.md?plain=1#L27

@snowman99, where is the "RESET button"?
Also, would you create an "Issues" section for the https://github.com/Keychron/zmk project?
Thanks.

ajs256, 2025-02-02:
I agree, it would be nice to have better-documented source for the B series. I know this might not be possible, but it would be awesome if it could be added into the main ZMK repo, so users can benefit from all the tooling built for "vanilla" ZMK.

dddping, 2025-02-12:
Hi, does anyone able to follow the guide at [https://github.com/Keychron/zmk/tree/keychron_bpro](url) to build a firmware?

I tried, but got error missing version.h file. The problem come from _app/src/dfu/tdfu_prv.h_, line 3
```
#ifndef __MY_DFU_PRIVATE_H__
#define __MY_DFU_PRIVATE_H__
#include "..\version.h"//<app_version.h>
unsigned char *my_dfu_get_prv_data(void);
unsigned char my_dfu_get_prv_len(void);

So, does it means the version.h missing under _app/src/dfu/_ ?

PeterMortensen, 2025-03-09:
Re *"follow the guide at"*: The actual link is to the QMK fork. The [correct link](https://github.com/Keychron/zmk/tree/keychron_bpro).

PeterMortensen, 2025-03-09:
Re *"error missing version.h file"*: I got that error too:

    FAILED: zephyr/CMakeFiles/zephyr.dir/home/embo/zmk_KeychronFork_forReal/app/src/dfu/tdfu_prv.c.obj 
    ccache /home/embo/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/embo/zmk_KeychronFork_forReal/zephyr/kernel/include -I/home/embo/zmk_KeychronFork_forReal/zephyr/arch/arm/include -I/home/embo/zmk_KeychronFork_forReal/zephyr/include -I/home/embo/zmk_KeychronFork_forReal/app/build/zephyr/include/generated -I/home/embo/zmk_KeychronFork_forReal/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/embo/zmk_KeychronFork_forReal/zephyr/soc/arm/nordic_nrf/common/. -I/home/embo/zmk_KeychronFork_forReal/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/nrfx_glue -I/home/embo/zmk_KeychronFork_forReal/zephyr/subsys/bluetooth -I/home/embo/zmk_KeychronFork_forReal/zephyr/subsys/usb/device -I/home/embo/zmk_KeychronFork_forReal/zephyr/subsys/settings/include -I/home/embo/zmk_KeychronFork_forReal/modules/hal/cmsis/CMSIS/Core/Include -I/home/embo/zmk_KeychronFork_forReal/modules/hal/nordic/nrfx -I/home/embo/zmk_KeychronFork_forReal/modules/hal/nordic/nrfx/drivers/include -I/home/embo/zmk_KeychronFork_forReal/modules/hal/nordic/nrfx/mdk -I/home/embo/zmk_KeychronFork_forReal/zephyr/modules/hal_nordic/nrfx/. -I/home/embo/zmk_KeychronFork_forReal/modules/crypto/tinycrypt/lib/include -I/home/embo/zmk_KeychronFork_forReal/app/module/include -I/home/embo/zmk_KeychronFork_forReal/app/module/drivers/sensor/battery/. -isystem /home/embo/zmk_KeychronFork_forReal/zephyr/lib/libc/minimal/include -isystem /home/embo/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/include -isystem /home/embo/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/include-fixed -fno-strict-aliasing -Os -imacros /home/embo/zmk_KeychronFork_forReal/app/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/home/embo/zephyr-sdk-0.16.3/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/embo/zmk_KeychronFork_forReal/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/embo/zmk_KeychronFork_forReal/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/embo/zmk_KeychronFork_forReal/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/embo/zmk_KeychronFork_forReal=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -Wfatal-errors -MD -MT zephyr/CMakeFiles/zephyr.dir/home/embo/zmk_KeychronFork_forReal/app/src/dfu/tdfu_prv.c.obj -MF zephyr/CMakeFiles/zephyr.dir/home/embo/zmk_KeychronFork_forReal/app/src/dfu/tdfu_prv.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/home/embo/zmk_KeychronFork_forReal/app/src/dfu/tdfu_prv.c.obj -c /home/embo/zmk_KeychronFork_forReal/app/src/dfu/tdfu_prv.c
    In file included from /home/embo/zmk_KeychronFork_forReal/app/src/dfu/tdfu_prv.c:1:
    /home/embo/zmk_KeychronFork_forReal/app/src/dfu/tdfu_prv.h:3:10: fatal error: ..\version.h: No such file or directory
        3 | #include "..\version.h"//<app_version.h>
          |          ^~~~~~~~~~~~~~
    compilation terminated.

Using:

    source ~/.ZMK_environment/bin/activate
    cd ~/zmk_KeychronFork_forReal/app
    west build --pristine  -b keychron -p  -- -DSHIELD=keychron_b6_uk
    deactivate

That was used a natively installed (not Docker) ZMK development environment. In the ZMK repository, building a sample keyboard worked:

    source ~/.ZMK_environment/bin/activate
    cd ~/zmk/app
    west build -b planck_rev6
    deactivate

[Transcript](https://pmortensen.eu/temp2/Transcript_of_building_B6_Pro_2025-03-08.txt).

PeterMortensen, 2025-03-09:
Why is there a backslash and two (forward) slashes in that line? Is that normal or expected?

PeterMortensen, 2025-03-10:
OK, changing that line to use the standard forward slash made it compile:

    /home/embo/zmk_KeychronFork_forReal/app/src/dfu/tdfu_prv.h:3:10: fatal error: ..\version.h: No such file or directory
    3 | #include "..\version.h"//<app_version.h>

The patch is required. Otherwise, the result is a linker error. Something like:

    /home/embo/zmk_KeychronFork_forReal/app/src/battery.c:90: undefined reference to `bt_bas_set_battery_level_fix'

PeterMortensen, 2025-03-10:
Re *"for official ZMK"*: That is a good idea.

What is story for ZMK? Is it different from QMK?

For example, [Keychron wrote](https://github.com/PeterMortensen/zmk_fork_of_Keychron_s_fork/blob/Copy_editing_of_the_build_instructions/README.md) (let's call it paraphrased):

> *"Keychron also added proprietary 2.4 GHz communication"*

[For QMK](https://wiki.keyboard.gay/KITS.html#claims-for-qmkvia-support-with-wireless), the most succinct explanation I have found is:

> *"the QMK maintainers refuse to merge support for boards that run QMK on one MCU and offload wireless connections to a separate chip."*

Would it be different for ZMK and this particular Keychron source code? (Not a rhetorical question; if it is not ultimately going to be accepted into the main ZMK repository it may not be worth doing)

bam80, 2025-03-10:
@PeterMortensen, wireless-friendliness (licensing including) was the main reason why ZMK exists.
Also, almost all ZMK-supported chips (B1 Pro's one is not an exception) have internal BT supported, so should be no any problem.

bam80, 2025-03-10:
> "the QMK maintainers refuse to merge support for boards that run QMK on one MCU and offload wireless connections to a separate chip."

All Keychron's ZMK keyboards have different single-SoC architecture I believe. Otherwise, there would be no point to base them on ZMK, probably.

PeterMortensen, 2025-03-10:
Re *"isn't something that can be resolved with a firmware update?"*: No, probably not. It is due to [the missing NKRO](https://www.reddit.com/r/Keychron/comments/1j77m44/comment/mgv3t8j/) (or 6 NKRO for that matter)

bam80, 2025-03-10:
> it consistently misses the final "e" when typing words like "node" quickly

Never experienced it myself but I saw Keychrone-specific de-bouncing code here, not sure if it's related.

PeterMortensen, 2025-03-14:
Re *"To build and flash the the firmware, hold down the RESET button"*: A workaround is to avoid wearing down the reset button is to add `&bootloader` to a key mapping somewhere

For example, to the same ***Fn*** + "***-***" ("***-***" on the ***Fn*** layer) that [stops working](https://github.com/Keychron/qmk_firmware/issues/344#issuecomment-2709384769) when using the firmware compiled from source. The only difference then is that it immediately switches to boot loader mode (without the 5 seconds time lock).

bam80, 2025-04-12:
Has anyone had any luck with Launcher?
#375

PeterMortensen, 2025-04-28:
Re *"create minimal PR for official ZMK"*: [A precedent](https://github.com/qmk/qmk_firmware/pull/24994) for the QMK-based [Lemokey X0](https://www.lemokey.com/products/lemokey-x0-qmk-wired-mechanical-gaming-keypad) macro pad/game pad.

Someone processed code from the Keychron fork and successfully brought it [into the main](https://github.com/qmk/qmk_firmware/tree/master/keyboards/keychron/x0) QMK repository (accepted on 2024-04-26).

bam80, 2025-04-28:
That's great, but I'm unaware of the similar precedents for ZMK yet.

closed 2025-10-06 by github-actions[bot]

00-kat, 2026-09-13:
> the source code will never be released

It was finally released → https://github.com/Keychron/zmk/commit/c284513085c005edf5f9a52b28c8090cc6ed01d2. I could get it to compile, but I haven't tested it yet.

bam80, 2026-09-13:
@00-kat Thanks, is it release for some newer B1 Pro variants which use non-ZMK-supported MCU (Realtek?) ?

00-kat, 2026-09-13:
It's one of the 2025 models (based on [Keychron's firmware flashing guide] I gather that there's two new versions; the released one seems to be for the latest one).

As far as I can tell from the source code, it's still using the same one as the original version (a [nRF52840]), but I don't have a newer one (yet?) so I can't tell for certain. It definitely compiles, and the binary blob for the 2.4 GHz one is definitely an nRF one, so I'm almost certain it's not using a Realtek/some other MCU.

PeterMortensen probably misinterpreted https://github.com/Keychron/qmk_firmware/issues/455#issuecomment-4358648495, which is about a *B11 Pro*, not a B1 Pro. Only B1, B2 and B6 Pro keyboards have ever had their source code released (probably because they're all the same keyboard in different sizes: 75% vs 96% vs 100% respectively); the *other* B series keyboards are probably the ones that use an unsupported microcontroller.

[nRF52840]: https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dk
[Keychron's firmware flashing guide]: https://www.keychron.com/blogs/archived/how-to-use-the-launcher-web-app-or-manually-flash-firmware-for-your-b-pro-series-keyboard

00-kat, 2026-09-13:
Oh a couple other things:

1. You need a Zephyr SDK 0.15.x for it to work (I'm using https://github.com/zephyrproject-rtos/sdk-ng/releases/download/v0.15.2/zephyr-sdk-0.15.2_linux-x86_64_minimal.tar.gz with the `arm-zephyr-eabi` toolchain installed and without host tools).
2. You may need to add `-DZEPHYR_TOOLCHAIN_VARIANT=zephyr`. (Maybe it's because I was using `-DZEPHYR_SDK_INSTALL_DIR="$(realpath zephyr-sdk-0.15.2)"` instead of installing the SDK in one of the [well-known directories].)
3. > A workaround for ZMK is to use a key mapping with key code `&bootloader`, for example, for the same ***Fn + '-'*** key; the only difference is that it will jump straight into the bootloader, without the time lock.

    From what I can tell, `&long_press_bootloader` defined in the keymap file is responsible for the time lock. For some reason `long_press_bootloader` is defined but unused, and the `-` key of `layer_three` (<kbd>fn</kbd> in Windows mode) is set to `&trans` (“fallback to previous layer”) instead of `&long_press_bootloader`; I feel like maybe setting that to `&long_press_bootloader` would make it work (but I haven't tested it yet; I'll comment here when I do).

    **Edit**: this is only the case for the keymap file of the original model (`b1`); the `b1n` ones have it set properly, so I feel like <kbd>fn</kbd>+<kbd>-</kbd> will Just Work™ on those.

[well-known directories]: https://docs.zephyrproject.org/4.1.0/develop/toolchains/zephyr_sdk.html#zephyr-sdk-installation