Keychron/qmk_firmware #345: [Feature Request] Set K5 Max RGB's Num Lock indicator to White to align with other Keychron keyboards

Piipperi, 2024-12-19:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Currently, the K5 Max RGB (possibly other K5 models too) sets the Num Lock key to Red. Most other Keychron RGB keyboards use White as indicator colors, and it would be nice if the K5 Max would do that as well, as White would blend-in better.

PeterMortensen, 2025-02-27:
Where is the difference expressed in [the source code](https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron)?

Piipperi, 2025-02-27:
To my understanding, Keychron most definitely has an internal branch they use, and then just push changes to the public one sometimes. 

It's entirely possible if I were to compile the firmware here, it could fix the numlock indicator color, but I am not sure if the other issues that were fixed on FW v1.0.1 would be on here (as the pre-compiled FW here is v1.0.0)

PeterMortensen, 2025-02-27:
Re *"Keychron most definitely has an internal branch they use"*: That is also my impression.

For example, holding ***F*** + ***J*** + ***Z*** for ***3 seconds*** to reset to factory defaults does not work for self-compiled software *as is*.

closed 2025-06-29 by github-actions[bot]