Keychron/qmk_firmware #348: [Bug] In K2 Pro, if RGB is off, the bluetooth connection resets every second in wireless mode

teddympq, 2024-12-26:
### Describe the Bug

Having RGB off causes the Bluetooth connection to disconnect every second of keyboard inactivity.

This ultimately causes DF(layer) return to default. Even if the default layer is changed to persistent, OSM mods sometimes won't work because the keyboard resets.

This issue never happens if:
- RGB is on
- RGB is off and the keyboard is in wireless mode but connected via cable
- RGB is off and the keyboard is in wired mode

Additionally, it won't happen if keys are pressed nonstop or keys are held.
I am using the latest firmware from the `wireless_playground` branch.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

PeterMortensen, 2024-12-27:
Related:

 * *[Keychron Q1 Max Bluetooth input lag and battery issues](https://www.reddit.com/r/Keychron/comments/1gkxgiv/keychron_q1_max_bluetooth_input_lag_and_battery/)*. E.g., *"When I turn off the LEDs (using* ***Fn*** *+* ***Tab),*** *the keyboard starts acting as described before. If I press ***Fn*** *+* ***Tab*** again to turn the LEDs back on, everything goes back to normal."*

teddympq, 2025-01-15:
I'm glad to see that they have found a solution. For the K Pro Series, the only workaround I can think of is creating a custom RGB layer with no LEDs activated

teddympq, 2025-01-23:
Hey! I think I figured out a possible solution! 

I tried to find a solution by tinkering a little bit... I noticed there was a function that activated if backlight is off

I tried making this function useless. As a result, the keyboard no longer resets, and works well with the backlight turned off. However, this function is vital for detecting whether the battery level is low and performing other necessary actions if needed. Is that correct? At least, now we know where the problem lies.
https://github.com/Keychron/qmk_firmware/blob/3fd4430bf27d08d20f91b231a04bcc92ed8b6771/keyboards/keychron/bluetooth/battery.c#L105-L140