Keychron/qmk_firmware #355: [Bug] Q3 Max: Waking keyboard from sleep on Bluetooth drops first keys typed

ottobonn, 2025-02-07:
### Describe the Bug

I have a new Q3 Max and I am using it over Bluetooth. After a while of inactivity, it goes to sleep. Pressing a key will wake it up, but it takes a few seconds to reconnect. 

During the time it takes the keyboard to wake up, any keys I press are ignored and not sent to the computer. Losing those key presses is annoying, because it's hard to know when the keyboard is ready for me to type. 

I would like to open a discussion on how we can avoid losing the keypresses as the keyboard wakes from sleep.

### Keyboard Used

Q3 Max

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q3-max-qmk-via-wireless-custom-mechanical-keyboard?variant=40766137565273

### Operating System

Linux Mint

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

ottobonn, 2025-02-07:
@lokher It looks like you're familiar with the relevant code branch. For what I gather, the Q3 Pro uses the `wireless_playground` branch? I am happy to help investigate this issue, but I want to make sure I am looking in the right place. 

First, it's worth asking if you can confirm this behavior. Do the initial key presses to wake the computer get dropped on the Q3 Max?

If this issue is due to how the firmware is written and not a hardware / OS issue, I think it would be worth investigating a fix. Here are some ideas that I wanted to ask if you have considered:

1. Queuing the key presses until reconnected to Bluetooth
2. Keeping Bluetooth connection active with HID over GATT
3. Changing the sleep mode to keep Bluetooth active

lokher, 2025-02-12:
I have tried adding key buffer before, the problem is bluetooth chip get a reset on wakeup from low power mode(in connection or disconnection), then sends a disconnection event to USB chip, and code clear the key buffer on a disconnection event.
I need to think more aobut it.

PeterMortensen, 2025-02-13:
I have a setup where [this does not happen](https://www.reddit.com/r/Keychron/comments/19fn0j3/comment/lc3lh1p/) (also Linux).

ottobonn, 2025-02-15:
@PeterMortensen just to make sure I understand, do you mean that, starting with the keyboard asleep from inactivity, you can begin typing, and it will connect to Bluetooth and capture all of the keys you have typed as it wakes up and connects? I have been testing it on various host operating systems, and in each test, with the keyboard asleep, it misses the first keypresses as it wakes up.

PeterMortensen, 2025-02-16:
Re *"starting with the keyboard asleep from inactivity you can begin typing, and it will connect to Bluetooth and capture all of the keys you have typed as it wakes up and connects?"*: Yes

Though I don't know if the keyboard is actually in sleep mode. Allegedly, it only goes to [real sleep after 2 hours](https://www.reddit.com/r/Keychron/comments/1ii2b8d/comment/mb6vpnf/) (of inactivity). But from a user perspective it ***definitely*** is (the RGB light is off, after the period set by `CONNECTED_BACKLIGHT_DISABLE_TIMEOUT`).

I have reduced the sleep time from 10 minutes to 30 seconds, because key strokes are never missed.

For instance, I started typing this sentence on a K5 Pro in sleep mode.

Sleep may or may not have to be prevented on the ***host side*** (of the Bluetooth adapter).

PeterMortensen, 2025-02-16:
What is your host side setup? For example, which Bluetooth adapter? What Bluetooth version does it support?