Keychron/qmk_firmware #357: [Feature Request]

Cenz0, 2025-02-15:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hi,Any news on the  source release of keychron K8 QMK ?
I can't wait :-)

PeterMortensen, 2025-02-19:
Can you make the title more specific, please?

Cenz0, 2025-02-19:
> Can you make the title more specific, please?

fixed

PeterMortensen, 2025-03-20:
[CDD](https://www.reddit.com/r/Keychron/comments/1ezuv5b/comment/lkpw7r5/) may or may not help.

BBaoVanC, 2025-03-20:
Just FYI that is "complaint driven development"... I had to follow 5 Reddit links deep on that

lokher, 2025-03-25:
Added.
https://github.com/Keychron/qmk_firmware/tree/wls_2025q1/keyboards/keychron/k8_version_2

closed 2025-03-26 by Cenz0