Keychron/qmk_firmware #358: [Bug]

juskek, 2025-02-24:
### Describe the Bug

**Context**
I'm trying to see print statements with `qmk console`
 **Replication Steps**
1. Added the following to my rules.mk file
```
// rules.mk
CONSOLE_ENABLE = yes   # Keep console debugging
MOUSEKEY_ENABLE = no   # Disable mouse movement with keyboard
EXTRAKEY_ENABLE = yes   # media/brightness keys (unless needed)
NKRO_ENABLE = no       # N-Key Rollover: unlimited number of simultaneous key presses
MIDI_ENABLE = no       # Turns keyboard into MIDI controller
SERIAL_ENABLE = no     # Allows communication between two QMK keyboards
STENO_ENABLE = no      # Stenography mode
```
2. `qmk compile -kb keychron/k3_max/ansi/rgb -km my_custom_keymap`
3. Then I see `error: #error There are not enough available endpoints to support all functions. Please disable one or more of the following: Mouse Keys, Extra Keys, Console, NKRO, MIDI, Serial, Steno`
<details open>
  <summary>Full Logs</summary>
 ```
Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=keychron/k3_max/ansi/rgb KEYMAP=my_custom_keymap KEYBOARD_FILESAFE=keychron_k3_max_ansi_rgb TARGET=keychron_k3_max_ansi_rgb_my_custom_keymap INTERMEDIATE_OUTPUT=.build/obj_keychron_k3_max_ansi_rgb_my_custom_keymap VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

⚠ keychron/k3_max/ansi/rgb: Feature mousekey is specified in both info.json and rules.mk, the rules.mk value wins.
⚠ keychron/k3_max/ansi/rgb: Feature extrakey is specified in both info.json and rules.mk, the rules.mk value wins.
⚠ keychron/k3_max/ansi/rgb: Feature nkro is specified in both info.json and rules.mk, the rules.mk value wins.
Generating: .build/obj_keychron_k3_max_ansi_rgb_my_custom_keymap/src/info_deps.d                    [OK]
⚠ keychron/k3_max/ansi/rgb: Feature mousekey is specified in both info.json and rules.mk, the rules.mk value wins.
⚠ keychron/k3_max/ansi/rgb: Feature extrakey is specified in both info.json and rules.mk, the rules.mk value wins.
⚠ keychron/k3_max/ansi/rgb: Feature nkro is specified in both info.json and rules.mk, the rules.mk value wins.
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_keychron_k3_max_ansi_rgb_my_custom_keymap/src/info_config.h                  [WARNINGS]
 | 
 | ⚠ keychron/k3_max/ansi/rgb: Feature mousekey is specified in both info.json and rules.mk, the rules.mk value wins.
 | ⚠ keychron/k3_max/ansi/rgb: Feature extrakey is specified in both info.json and rules.mk, the rules.mk value wins.
 | ⚠ keychron/k3_max/ansi/rgb: Feature nkro is specified in both info.json and rules.mk, the rules.mk value wins.
 | 
Generating: .build/obj_keychron_k3_max_ansi_rgb_my_custom_keymap/src/default_keyboard.c             [WARNINGS]
 | 
 | ⚠ keychron/k3_max/ansi/rgb: Feature mousekey is specified in both info.json and rules.mk, the rules.mk value wins.
 | ⚠ keychron/k3_max/ansi/rgb: Feature extrakey is specified in both info.json and rules.mk, the rules.mk value wins.
 | ⚠ keychron/k3_max/ansi/rgb: Feature nkro is specified in both info.json and rules.mk, the rules.mk value wins.
 | 
Generating: .build/obj_keychron_k3_max_ansi_rgb_my_custom_keymap/src/default_keyboard.h             [WARNINGS]
 | 
 | ⚠ keychron/k3_max/ansi/rgb: Feature mousekey is specified in both info.json and rules.mk, the rules.mk value wins.
 | ⚠ keychron/k3_max/ansi/rgb: Feature extrakey is specified in both info.json and rules.mk, the rules.mk value wins.
 | ⚠ keychron/k3_max/ansi/rgb: Feature nkro is specified in both info.json and rules.mk, the rules.mk value wins.
 | 
Compiling: ./keyboards/keychron/common/wireless/wireless.c                                          [OK]
Compiling: ./keyboards/keychron/common/wireless/report_buffer.c                                     [OK]
Compiling: ./keyboards/keychron/common/wireless/lkbt51.c                                            [OK]
Compiling: ./keyboards/keychron/common/wireless/indicator.c                                        In file included from tmk_core/protocol/chibios/usb_main.h:15,
                 from ./keyboards/keychron/common/wireless/indicator.c:26:
tmk_core/protocol/usb_descriptor.h:285:6: error: #error There are not enough available endpoints to support all functions. Please disable one or more of the following: Mouse Keys, Extra Keys, Console, NKRO, MIDI, Serial, Steno
  285 | #    error There are not enough available endpoints to support all functions. Please disable one or more of the following: Mouse Keys, Extra Keys, Console, NKRO, MIDI, Serial, Steno
      |      ^~~~~
 [ERRORS]
 | 
 | 
 | 
gmake: *** [builddefs/common_rules.mk:373: .build/obj_keychron_k3_max_ansi_rgb_my_custom_keymap/common/wireless/indicator.o] Error 1
/Users/justinkek/Desktop/Repositories/personal/qmk/qmk_firmware/lib/python/qmk/decorators.py:20: UserWarning: cli._subcommand has been deprecated, please use cli.subcommand_name to get the subcommand name instead.
  if cli.config_source[cli._subcommand.__name__]['keyboard'] != 'argument':
/Users/justinkek/Desktop/Repositories/personal/qmk/qmk_firmware/lib/python/qmk/decorators.py:40: UserWarning: cli._subcommand has been deprecated, please use cli.subcommand_name to get the subcommand name instead.
  if cli.config_source[cli._subcommand.__name__]['keymap'] != 'argument':
Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s flash KEYBOARD=keychron/k3_max/ansi/rgb KEYMAP=my_custom_keymap KEYBOARD_FILESAFE=keychron_k3_max_ansi_rgb TARGET=keychron_k3_max_ansi_rgb_my_custom_keymap INTERMEDIATE_OUTPUT=.build/obj_keychron_k3_max_ansi_rgb_my_custom_keymap VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

⚠ keychron/k3_max/ansi/rgb: Feature mousekey is specified in both info.json and rules.mk, the rules.mk value wins.
⚠ keychron/k3_max/ansi/rgb: Feature extrakey is specified in both info.json and rules.mk, the rules.mk value wins.
⚠ keychron/k3_max/ansi/rgb: Feature nkro is specified in both info.json and rules.mk, the rules.mk value wins.
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: ./keyboards/keychron/common/wireless/indicator.c                                        In file included from tmk_core/protocol/chibios/usb_main.h:15,
                 from ./keyboards/keychron/common/wireless/indicator.c:26:
tmk_core/protocol/usb_descriptor.h:285:6: error: #error There are not enough available endpoints to support all functions. Please disable one or more of the following: Mouse Keys, Extra Keys, Console, NKRO, MIDI, Serial, Steno
  285 | #    error There are not enough available endpoints to support all functions. Please disable one or more of the following: Mouse Keys, Extra Keys, Console, NKRO, MIDI, Serial, Steno
      |      ^~~~~
 [ERRORS]
 | 
 | 
 | 
gmake: *** [builddefs/common_rules.mk:373: .build/obj_keychron_k3_max_ansi_rgb_my_custom_keymap/common/wireless/indicator.o] Error 1
```
</details>

### Keyboard Used

keychron/k3_max/ansi/rgb

### Operating System

macOS

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

PeterMortensen, 2025-02-27:
Can you make the title more specific, please?

lokher, 2025-03-05:
`error: #error There are not enough available endpoints to support all functions. Please disable one or more of the following: Mouse Keys, Extra Keys, Console, NKRO, MIDI, Serial, Steno`

the USB resouce/capability of this chip is limited, the firmware use all the endpoints, and console feature need an endpoint, you have to disable NKRO, Extra key and mouse key feature to release an endpoint.