Keychron/qmk_firmware #360: [Feature Request] Use backlight to indicate mode when toggling NKRO

hexpunk, 2025-02-27:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

### Description

When NKRO is toggled using a hotkey combination, the keyboard does not give any indication as to whether you're toggling it on or off.

I have tested the keyboard at https://www.microsoft.com/applied-sciences/projects/anti-ghosting-demo and NKRO is correctly being toggled.

The issue is just that the keyboard gives no indication as to what the mode you've toggled to.  I'd love it if a key could light up green for on and red for off.  But that's just a suggestion.  If the keyboard could indicate what mode it is in when NKRO is toggled, that would be a huge improvement.

hexpunk, 2025-02-27:
FWIW, the battery check hotkey already uses backlights to show the battery status, so there is precedent for such a feature.

PeterMortensen, 2025-03-01:
I support this feature request, but it is an undocumented feature (for whatever reason). Keychron didn't document it either for the original (non-QMK) K series with proprietary firmware (though [it did slip out](https://web.archive.org/web/20201215083102/https://www.keychron.com/blogs/news/difference-among-keychron-keyboards) that there was such a setting).

hexpunk, 2025-03-01:
FWIW, my keyboard is a Q60 Max. The battery check hotkey and NKRO toggle hotkey were mapped by default to fn+b and fn+n respectively. I was able to test NKRO on my board in all three modes: bluetooth, wired, and "g" wireless USB dongle. It works as expected in all modes.

I just can't tell if NKRO is on or not without going to a testing web app and smashing keys, which is less than ideal.

PeterMortensen, 2025-03-01:
Re *"I just can't tell if NKRO is on or not without going to a testing web app"*: [This test](https://www.reddit.com/r/Keychron/comments/14mla2x/comment/l0wsy3a/) may be simpler.

Though it is still a workaround.

antonlvovych, 2025-11-15:
What I did - I enabled it and removed that hotkey from the mappings so I won’t accidentally turn it off again 😅