Keychron/qmk_firmware #363: [Bug] keychron/q6_max/ansi_encoder fails to compile with "rgb_matrix" : false

donnoman, 2025-03-14:
### Describe the Bug

Git Branch: wireless_playground pulled 3/12/25
relevant info.json
```
    "features": {
        "bootmagic": false,
        "extrakey" : false,
        "mousekey" : false,
        "dip_switch" : true, see #362 
        "encoder": false,
        "encoder_map": false,
        "nkro" : false,
        "rgb_matrix": false, 
        "raw" : false,
        "send_string" : false
    },
```
compile output (partial) 
```
Compiling: ./keyboards/keychron/common/wireless/lpm.c                                              In file included from ./lib/chibios/os/hal/include/hal_pal.h:174,
                 from ./lib/chibios/os/hal/include/hal.h:306,
                 from ./platforms/chibios/drivers/wear_leveling/wear_leveling_efl_config.h:6,
                 from <command-line>:
./keyboards/keychron/common/wireless/lpm.c: In function 'pre_enter_low_power_mode':
./keyboards/keychron/common/wireless/lpm.c:165:20: error: 'SPI_SCK_PIN' undeclared (first use in this function)
  165 |     palSetLineMode(SPI_SCK_PIN, PAL_MODE_INPUT_PULLDOWN);
      |                    ^~~~~~~~~~~
```
...
```
./keyboards/keychron/common/wireless/lpm.c:192:5: note: in expansion of macro 'palSetLineMode'
  192 |     palSetLineMode(SPI_MOSI_PIN, PAL_MODE_ALTERNATE(5));
      |     ^~~~~~~~~~~~~~
 [ERRORS]
 |
 |
 |
make: *** [builddefs/common_rules.mk:373: .build/obj_keychron_q6_max_ansi_encoder_via/common/wireless/lpm.o] Error 1
```

### Keyboard Used

keychron/q6_max/ansi_encoder

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q6-max-qmk-via-wireless-custom-mechanical-keyboard

### Operating System

Windows 10

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: C:/Users/donovan/qmk_firmware
Ψ Detected Windows 10 (10.0.19045).
Ψ QMK MSYS version: 1.10.0
Ψ Userspace enabled: False
Ψ Git branch: wireless_playground
Ψ Repo version: 0.14.29
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.3.0
Ψ Found avr-gcc version 14.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.3-2024041
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-01-10 10:36:30 +0800 --  (86a3ed9baf)
Ψ - lib/chibios-contrib: 2024-06-04 11:39:47 +0800 --  (5dbc2a07)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

The reason I'm trying to compile and messing with features is because the q6_max fails to work on my iogear kvm. In it's factory state the USB bus recycles, it kills the mouse and the keyboard. Neither work with the keyboard on the kvm. Only way to get anything out of the keyboard is to hook it up direct.

disabling NKRO was the first thing I tried, which changed nothing. Which I was trying to disable all features to get the board to it's absolute minimal state, to see if I could get it to work with my KVM.

KVM works with other keyboards including my Montech Mkey.

PeterMortensen, 2025-03-20:
I can confirm that. ***Only*** changing the single line in file *[info.json](https://github.com/Keychron/qmk_firmware/blob/wireless_playground/keyboards/keychron/q6_max/info.json#L19)*  to

    "rgb_matrix": false,

resulted in the compile error.

Command line:

    qmk clean # To make changes (if any) to
              # .json files take effect
    qmk compile -kb keychron/q6_max/ansi_encoder -km via

Some of the compile errors:

    Compiling: ./keyboards/keychron/common/wireless/lpm.c                                              In file included from ./lib/chibios/os/hal/include/hal_pal.h:174,
                     from ./lib/chibios/os/hal/include/hal.h:306,
                     from ./platforms/chibios/drivers/wear_leveling/wear_leveling_efl_config.h:6,
                     from <command-line>:
    ./keyboards/keychron/common/wireless/lpm.c: In function 'pre_enter_low_power_mode':
    ./keyboards/keychron/common/wireless/lpm.c:165:20: error: 'SPI_SCK_PIN' undeclared (first use in this function)
      165 |     palSetLineMode(SPI_SCK_PIN, PAL_MODE_INPUT_PULLDOWN);
          |                    ^~~~~~~~~~~
    ./lib/chibios/os/hal/ports/STM32/LLD/GPIOv2/hal_pal_lld.h:415:25: note: in definition of macro 'pal_lld_setgroupmode'
      415 |   _pal_lld_setgroupmode(port, mask << offset, mode)
          |                         ^~~~
    ./lib/chibios/os/hal/include/hal_pal.h:698:3: note: in expansion of macro 'palSetGroupMode'
      698 |   palSetGroupMode(port, PAL_PORT_BIT(pad), 0U, mode)
          |   ^~~~~~~~~~~~~~~

PeterMortensen, 2025-03-25:
Supposedly fixed with [3A4459](https://github.com/Keychron/qmk_firmware/commit/3a445969c471a26def3d5a7fa540ca756e305965) (2025-03-25).