Keychron/qmk_firmware #364: [Feature Request] Lemokey X3 RGB source code release

fIREmiNeR1, 2025-03-15:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hello, is Lemokey X3 RGB firmware going to be released? If yes, is there any approximate date?

PeterMortensen, 2025-03-28:
There is [some code](https://github.com/Keychron/qmk_firmware/tree/playground/keyboards/keychron/x3) in branch ***playground***.

PeterMortensen, 2025-03-28:
You best bet may be [CDD](https://www.reddit.com/r/Keychron/comments/1jh79da/comment/mj61jfe).

PeterMortensen, 2025-08-20:
Now something  *[happened in ***playground***](https://github.com/Keychron/qmk_firmware/commits/playground/) (2025-08-21):

> *"[Added X3 ANSI RGB, ISO Red/RGB](https://github.com/Keychron/qmk_firmware/commit/4a8143cce9c2ab32a1384c352c3db324149abddd)"*

fIREmiNeR1, 2025-08-22:
Finally this happened. I have already lost all hopes that they will do something. Thanks for help!