Keychron/qmk_firmware #367: [Bug] Q15 Max, set_single_persistent_default_layer not persistent

sbmueller, 2025-03-22:
### Describe the Bug

Hi, I'm tyring to implement persistent keyboard layers on my Q15. I wrote this function in my custom keymap:
```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case _QWERTY:
            if (record->event.pressed) {
                set_single_persistent_default_layer(QWERTY);
            }
            return false;
            break;
        case _COLEMAK_DH:
            if (record->event.pressed) {
                set_single_persistent_default_layer(COLEMAK_DH);
            }
            return false;
            break;
    }
    if (!process_record_keychron_common(keycode, record)) {
        return false;
    }
    return true;
}

The switch of layouts also works like this during runtime. But after I reboot the keyboard, the layer is not persistent and again I end up in layer 0 (QWERTY).
My expectation is that even after I reboot the keyboard, the last layer is persistent and active.

```

### Keyboard Used

Q15 Max

### Link to product page (if applicable)

https://www.keychron.com/products/keychron-q15-max-qmk-wireless-custom-mechanical-keyboard

### Operating System

MacOS

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: /Users/senpo/src/qmk_firmware
Ψ Detected macOS 15.3.2 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: sbmueller-wireless
Ψ Repo version: 0.23.1
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest sbmueller-wireless: 2025-02-02 16:01:55 +0100 (507b4ca615) -- Q15: Add keymap
Ψ - Latest upstream/master: 2025-01-29 11:28:58 -0800 (fa98117a3e) -- Add Keychron C3 Pro (Red & RGB) (#24874)
Ψ - Latest upstream/develop: 2025-01-29 19:30:22 +0000 (ff09b921f1) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2023-12-18 13:53:02 +1100 (9539f135d8) -- Remove obvious user keymaps, `keyboards/[0-9]*` edition. (#22691)
Ψ - Common ancestor with upstream/develop: 2023-12-18 13:53:02 +1100 (9539f135d8) -- Remove obvious user keymaps, `keyboards/[0-9]*` edition. (#22691)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-01-10 10:36:30 +0800 --  (86a3ed9baf)
Ψ - lib/chibios-contrib: 2024-06-04 11:39:47 +0800 --  (5dbc2a07)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

The full keymap is [here](https://github.com/sbmueller/qmk_firmware/blob/bb799c84f94453db2c80ec4978de78fa6b13dafc/keyboards/keychron/q15_max/ansi_encoder/keymaps/sbmueller/keymap.c).

lokher, 2025-03-24:
The default layer is load but set by dip switch after keyboard rebooting：
https://github.com/Keychron/qmk_firmware/blob/3fd4430bf27d08d20f91b231a04bcc92ed8b6771/keyboards/keychron/q15_max/q15_max.c#L33

Remove below lines to disable DIP switch should works, but OS switch of keyboard won't work anymore.  https://github.com/Keychron/qmk_firmware/blob/3fd4430bf27d08d20f91b231a04bcc92ed8b6771/keyboards/keychron/q15_max/info.json#L28-L30

sbmueller, 2025-03-24:
@lokher oh, thanks! I didn't make the connection that the OS switch is selecting the default layer. Thanks!

closed 2025-03-24 by sbmueller