Keychron/qmk_firmware #368: [Feature Request] k3_pro qmk-firmware update from 0.19.2 to 0.28

kitsunekyo, 2025-03-28:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

i've notice that the latest tag on the bluetooth_playground branch is massively outdated. can this be updated?

PeterMortensen, 2025-03-28:
Can you elaborate?

lokher, 2025-04-01:
Try this branch:
https://github.com/Keychron/qmk_firmware/tree/wls_2025q1

kitsunekyo, 2025-04-04:
Generally I'd really like to have my "qmk supported" keyboard to support all current qmk features. The version numbers are basically just taken from a comparison of the current official qmk version to keychrons fork.

The current version seems to be missing features like the [`CHORDIAL_HOLD`](https://docs.qmk.fm/tap_hold#chordal-hold). I also noticed some other oddities like `HOLD_ON_OTHER_KEYPRESS` to be enabled by default, so I had to implement [this weird hack](https://github.com/kitsunekyo/qmk_k3_pro_neo_layer3/blob/main/keymap.c#L125), so my homerow mods are even usable.

Thank you @lokher for the link to the branch. Can you explain how this branch differs from the `bluetooth_playground` branch?

I'm really thankful for you sharing that branch, but I struggle with understanding your branching structure or how I as a customer am supposed to find any of this.

kitsunekyo, 2025-04-04:
Unfortunately it does not build for me at all. [SOURCE](https://github.com/kitsunekyo/qmk_k3_pro_neo_layer3)

```
❯ qmk compile -kb keychron/k3_pro/iso/rgb -km kitsunekyo
Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=keychron/k3_pro/iso/rgb KEYMAP=kitsunekyo KEYBOARD_FILESAFE=keychron_k3_pro_iso_rgb TARGET=keychron_k3_pro_iso_rgb_kitsunekyo VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

Generating: .build/obj_keychron_k3_pro_iso_rgb_kitsunekyo/src/info_deps.d                           [OK]
arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   65814       0   65814   10116 keychron_k3_pro_iso_rgb_kitsunekyo.bin

Generating: .build/obj_keychron_k3_pro_iso_rgb_kitsunekyo/src/community_modules.h                   [OK]
Generating: .build/obj_keychron_k3_pro_iso_rgb_kitsunekyo/src/community_modules.c                   [OK]
Generating: .build/obj_keychron_k3_pro_iso_rgb_kitsunekyo/src/community_modules_introspection.c     [OK]
Generating: .build/obj_keychron_k3_pro_iso_rgb_kitsunekyo/src/community_modules_introspection.h     [OK]
Generating: .build/obj_keychron_k3_pro_iso_rgb_kitsunekyo/src/info_config.h                         [OK]
Generating: .build/obj_keychron_k3_pro_iso_rgb_kitsunekyo/src/default_keyboard.c                    [OK]
Generating: .build/obj_keychron_k3_pro_iso_rgb_kitsunekyo/src/default_keyboard.h                    [OK]
Compiling: keyboards/keychron/k3_pro/matrix.c                                                      In file included from ./lib/chibios/os/hal/osal/rt-nil/osal.h:32,
                 from ./lib/chibios/os/hal/include/hal.h:28,
                 from ./platforms/chibios/drivers/wear_leveling/wear_leveling_efl_config.h:6,
                 from <command-line>:
./lib/chibios/os/rt/include/ch.h:125:10: fatal error: chlib.h: No such file or directory
  125 | #include "chlib.h"
      |          ^~~~~~~~~
compilation terminated.
 [ERRORS]
 |
 |
 |
gmake: *** [builddefs/common_rules.mk:362: .build/obj_keychron_k3_pro_iso_rgb_kitsunekyo/matrix.o] Error 1
```

kitsunekyo, 2025-04-04:
Thank you. With this branch, `CHORDIAL_HOLD` seems to work as expected. I would still love to understand what this newest branch is now, and how we could improve discoverability for other users.

42Craft, 2025-04-05:
Interesting branch. The name suggests it contains fixes made in 2025 q1.

PeterMortensen, 2025-06-03:
Re *"The name suggests it contains fixes made in 2025 q1"*: It seems to be based on QMK as of 2025-03-03 ([F0A5DB](https://github.com/qmk/qmk_firmware/commit/F0A5DB96C8A128BDF7F8321FDC1E0011160AEB3A)).

The QMK [0.28.0](https://docs.qmk.fm/ChangeLog/20250223) release was on 2025-02-23.

The QMK [0.29.0](https://docs.qmk.fm/ChangeLog/20250223) release was on 2025-05-25.