Keychron/qmk_firmware #369: [Feature Request] Upstreeam to QMK

Henry-Hiles, 2025-03-28:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hello, I'd like to see some of this firmware upstreamed to qmk, like the Keychron V1 Max. The V1 Max hasn't been updated in 2 months, so it's obviously fairly stable. Thanks.

PeterMortensen, 2025-03-28:
Related:

 * *[The current software state](https://github.com/Keychron/qmk_firmware/issues/217)* (numero 217. 2024-01-16)

lokher, 2025-04-01:
Try this branch:
https://github.com/Keychron/qmk_firmware/tree/wls_2025q1

Henry-Hiles, 2025-04-01:
> Try this branch: https://github.com/Keychron/qmk_firmware/tree/wls_2025q1

Trying rn

Henry-Hiles, 2025-04-01:
This doesn't seem right, haha:
```
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s flash KEYBOARD=keychron/v1_max/ansi_encoder KEYMAP=default KEYBOARD_FILESAFE=keychron_v1_max_ansi_encoder TARGET=keychron_v1_max_ansi_encoder_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk" QMK_USERSPACE=/home/quadradical/Documents/Code/qmk_userspace

Generating: .build/obj_keychron_v1_max_ansi_encoder_default/src/info_deps.d                         [OK]
arm-none-eabi-gcc (Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35)) 12.3.1 20230626
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_keychron_v1_max_ansi_encoder_default/src/community_modules.h                 [OK]
Generating: .build/obj_keychron_v1_max_ansi_encoder_default/src/community_modules.c                 [OK]
Generating: .build/obj_keychron_v1_max_ansi_encoder_default/src/community_modules_introspection.c   [OK]
Generating: .build/obj_keychron_v1_max_ansi_encoder_default/src/community_modules_introspection.h   [OK]
Generating: .build/obj_keychron_v1_max_ansi_encoder_default/src/info_config.h                       [OK]
Generating: .build/obj_keychron_v1_max_ansi_encoder_default/src/default_keyboard.c                  [OK]
Generating: .build/obj_keychron_v1_max_ansi_encoder_default/src/default_keyboard.h                  [OK]
Compiling: ./keyboards/keychron/common/wireless/wireless.c                                          [OK]
Compiling: ./keyboards/keychron/common/wireless/report_buffer.c                                     [OK]
Compiling: ./keyboards/keychron/common/wireless/lkbt51.c                                            [OK]
Compiling: ./keyboards/keychron/common/wireless/indicator.c                                         [OK]
Compiling: ./keyboards/keychron/common/wireless/wireless_main.c                                     [OK]
Compiling: ./keyboards/keychron/common/wireless/transport.c                                         [OK]
Compiling: ./keyboards/keychron/common/wireless/lpm.c                                              ./keyboards/keychron/common/wireless/lpm.c: In function 'lpm_task':
./keyboards/keychron/common/wireless/lpm.c:293:14: error: implicit declaration of function 'bat_level_animiation_actived' [-Werror=implicit-function-declaration]
  293 |             !bat_level_animiation_actived()
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make: *** [builddefs/common_rules.mk:362: .build/obj_keychron_v1_max_ansi_encoder_default/common/wireless/lpm.o] Error 1
```

Henry-Hiles, 2025-04-01:
Doesn't build.

lokher, 2025-04-02:
Did you change the code?
RGB_MATRIX_ENABLE is enabled, so the line mentioned in the error should not
be compiled.

[image: image.png]

Henry-Hiles, 2025-04-02:
> Did you change the code?
> RGB_MATRIX_ENABLE is enabled, so the line mentioned in the error should not
> be compiled.
> 
> [image: image.png]
> […](#)

Oh, I have RGB disabled, do I need it? I used the config.mk or whatever

lokher, 2025-04-02:
RGB matrix is required for bluetooth/2.4g pairing or reconnecting indication.

Henry-Hiles, 2025-04-02:
> RGB matrix is required for bluetooth/2.4g pairing or reconnecting indication.

Really? It seems to still work without, it just doesn't give indicators. Is there a better way to disable just the RGB effects?

lokher, 2025-04-03:
Change the `if` condition to below:
```c
        if (1
#ifdef LED_MATRIX_ENABLE
            (!led_matrix_is_enabled() ||
            (led_matrix_is_enabled() && led_matrix_is_driver_shutdown()))
#elif defined(RGB_MATRIX_ENABLE)
            (!rgb_matrix_is_enabled() ||
            (rgb_matrix_is_enabled() && rgb_matrix_is_driver_shutdown()))
#else
          // !bat_level_animiation_actived()
#endif
            && !lpm_any_matrix_action()) {
```

closed 2025-08-02 by github-actions[bot]

Henry-Hiles, 2025-08-02:
Still valid (I do hate stale bots...)