Keychron/qmk_firmware #373: [Feature Request] QMK Community Modules

42Craft, 2025-04-06:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Dear Keychron Team,

I'm interested in using the recently released (Q1 2025 update) QMK Community Modules with my Keychron keyboard. I've reviewed the official QMK documentation on this feature: https://docs.qmk.fm/features/community_modules.

Are Community Modules fully compatible with all Keychron QMK-based keyboards? I own a Keychron k15 pro and k11 max. What are the steps to utilize these new features.

L0nelyE4rth, 2025-06-18:
It's included in [2025q1] (https://github.com/Keychron/qmk_firmware/tree/wls_2025q1). You can try from that branch with your module.

poiu477, 2025-06-18:
Thanks bud