Keychron/qmk_firmware #374: Post the source code to currently shipping K4 Pro firmware

jzw95, 2025-04-11:
### Issue Description

I have compiled and flashed the firmware for the K4 Pro that's in the `bluetooth_playground` branch of the `qmk_firmware` repository. The keyboard works fine, but this is clearly not the same firmware that shipped with the keyboard and is available in binary form from [the Keychron firmware download page](https://www.keychron.com/pages/firmware-and-json-files-of-the-keychron-qmk-k-pro-and-k-max-series-keyboards). The shipping firmware includes `MCtl` and `LPad` in the custom/user key enumeration in `k4_pro.h`. I can see this because when the keyboard is running the shipping firmware and I upload the `k4_pro_iso_rgb_v1.00.json` file (also provided on the Keychron firmware download page) to VIA, the Mission Control & Launchpad keys are labelled correctly, as are the BTH1, BTH2, BTH3, and Batt keys on layer 1. The `k4_pro_iso_rgb_v1.00.json` file from the firmware page includes definitions for 15 `customKeycodes`, including `MCtl` and `LPad`. So it's clearly matching up with the firmware also having enumerated 15 custom keys in `k4_pro.h`.

On the other hand, when using firmware compiled from this repository (and precompiled in [`keyboards/keychron/k4_pro/firmware/keychron_k4_pro_iso_rgb_via.bin`](keyboards/keychron/k4_pro/firmware/keychron_k4_pro_iso_rgb_via.bin)), those keys are mislabelled in both VIA and the Keychron Launcher. The [`k4_pro.h`](https://github.com/Keychron/qmk_firmware/blob/bluetooth_playground/keyboards/keychron/k4_pro/k4_pro.h) provided here only enumerates 13 custom keys, not including `MCtl` and `LPad`. The VIA keymap in the repository ([`keyboards/keychron/k4_pro/via_json/k4_pro_iso_rgb.json`](keyboards/keychron/k4_pro/via_json/k4_pro_iso_rgb.json)) matches this and only defines 13 `customKeycodes`.

Fixing this in the provided source code (adding to the enum declaration, putting the relevant new keycodes in `k4_pro.c`, etc) is straightforward, but my concern is that the source code provided here isn't the same that's used to build the shipping firmware. So there may be other fixes in the shipping firmware. As I customise the firmware, I want to use the most up-to-date, bug-fixed starting point. So could you upload the source code for the currently-shipping firmware?

On a related note, is `bluetooth_playground` the correct branch to be using? I'm confused as to which branch is the proper one with the most up-to-date code. I did look at the `wireless_playground` branch but it has the same issue with not including the `MCtl` and `LPad` enums, and since the K4 Pro page on Keychron's website links to the `bluetooth_playground` branch I have stuck with that for now.

42Craft, 2025-04-11:
Your experience reminds me of a similar experience I had with the K11 Max.

I flashed the firmware from the wireless playground branch. After that, the short keys to show battery stopped working.

PeterMortensen, 2025-04-11:
Perhaps the effective values are in [common/keychron_common.h](https://github.com/Keychron/qmk_firmware/blob/wireless_playground/keyboards/keychron/common/keychron_common.h#L27)?

It includes `KC_MCTRL` and `KC_LNPAD`.

jzw95, 2025-04-12:
Quick update: [common/keychron_common.h](https://github.com/Keychron/qmk_firmware/blob/wireless_playground/keyboards/keychron/common/keychron_common.h) and [common/keychron_common.c](https://github.com/Keychron/qmk_firmware/blob/wireless_playground/keyboards/keychron/common/keychron_common.c) are not included when compiling firmware for the K4 Pro in the wireless playground repository. Therefore the custom keycode enum doesn't include `KC_MCTRL` and `KC_LNPAD`. So this still isn't the code used to build the shipping firmware. It may be close to it, but as it exists in the repository it doesn't compile to the same binary as the firmware Keychron provides with its keyboard or on its website.