Keychron/qmk_firmware #375: Keychron Launcher: "No Valid HID Device Configuration Found"

bam80, 2025-04-12:
### Issue Description

The browser sees Keychron Receiver but get the error on Connect:
https://launcher.keychron.com/

I followed udev rules advise here but the problem is seems with something different:  
https://www.reddit.com/r/Keychron/comments/1e5um1u/a_linux_user_psa/

Keychron B1 Pro , Chromium, Fedora 41 Silverblue.

bam80, 2025-04-12:
Got it working, but on wired connection only:
https://keychronsupport.zendesk.com/hc/en-us/articles/22776472488855-My-keyboard-is-not-recognized-by-Keychron-Launcher-what-should-I-do

Whereas instruction on https://launcher.keychron.com/ clearly says
it should be connectable by 2.4GHz:
[image]

PeterMortensen, 2025-04-14:
Re *"Please connect your device to your computer using a cable or 2.4 GHz"*: They may have become confused by their [own messages](https://www.reddit.com/r/Keychron/comments/1jubjv5/comment/mm0zv0b).

bam80, 2025-07-17:
@PeterMortensen the link gives "Unauthorized access" for me.

PeterMortensen, 2025-07-31:
Sorry, I used the wrong link. It has been corrected.