Keychron/qmk_firmware #381: Are 2.4GHz connections enrypted?

Gabriel85, 2025-05-04:
### Issue Description

I don't see any code regarding encryption of the 2.4GHz connection with the dongle. Are those encrypted?
I know the micro-controller supports it, but I can't find any code related to it. If so, can somebody please point me to it? Thanks!

PeterMortensen, 2025-05-04:
How is encryption supposed to work? E.g., is it a feature or mode of the 2.4 GHz module?

PeterMortensen, 2026-02-21:
Re *"...said "Added AES128 encryption for wireless keyboard keys" in the changelog"*: That is on [this page](https://www.keychron.com/pages/how-to-flash-the-firmware-for-the-keychron-receiver) for version 3.0 of the '2.4&nbsp;[GHz](https://en.wikipedia.org/wiki/Hertz#SI_multiples)' dongle (*"Firmware version d.3.0/c.3.0 updated on July 9, 2024"*):

> *"AES128 encryption for wireless keyboard data."*