Keychron/qmk_firmware #384: [Bug] V1 Max Firmware on Branch - Firmware Version Outdated

ProgrammerManSudjo, 2025-05-17:
### Describe the Bug

I compiled my own `keymap.c`, `rules.mk` and `config.h` files. When launching on `Keychron Launcher` it still prompts me for a firmware update.

It mentions that my current firmware version is `1.0.1`, even though I pulled from this branch (the latest updated) for these updates.

Mainly I'm missing on LKP, Per Key RGB, Debounce, Sleep Time Adjustments and Advanced Controls.

Is there a way I can fix this, or is the branch not yet updated to the latest firmware's features?

[image]

### Keyboard Used

keychron/v1_max

### Operating System

Windows 11

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: C:/Users/Sudjo/qmk_firmware
Ψ Detected Windows 11 (10.0.22631).
Ψ QMK MSYS version: 1.10.0
Ψ Userspace enabled: False
Ψ Git branch: wls_2025q1
Ψ Repo version: 0.14.29
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.3.0
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 14.2.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 7.3-2024041
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2025-03-14 17:30:23 +0800 --  (bc63876dc)
Ψ - lib/chibios-contrib: 2025-03-20 16:21:08 +0800 --  (02e126c5)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [x] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

no

PeterMortensen, 2025-05-29:
This line in file *info.json* has been there since the inception of that branch (early 2024):

    "build": {
        "debounce_type": "sym_eager_pk"
    },
    "debounce": 20

It it nothing new.

PeterMortensen, 2025-06-03:
The *"2025q1"* in the branch name "wls_2025q1" [may refer to](https://www.reddit.com/r/Keychron/comments/1koojfu/comment/mvrn4t2/) the ***QMK version***, and ***not*** [the early 2025 Keychron keyboard main firmware updates](https://www.reddit.com/r/Keychron/comments/1j092s2/comment/ml0wljn/).

PeterMortensen, 2025-06-13:
Re *"I'm missing on [LKP](https://www.reddit.com/r/Keychron/comments/1eh4qog/comment/lfyu7zf/), per-key RGB, debounce, sleep time adjustments and advanced controls"*: Those [were released](https://github.com/Keychron/qmk_firmware/issues/217#issuecomment-2937073410) on 2025-05-30 into the ***"[wireless_playground](https://github.com/Keychron/qmk_firmware/tree/wireless_playground/keyboards/keychron)"**** branch, not "wls_2025q1" (the 2025 in "wls_2025q1" refers to the QMK version, not that Keychron's software).

Though it *seems* to be a partial release.

ProgrammerManSudjo, 2025-06-14:
I saw the updates, they added the new features for V3 Max. Hopefully soon they'll release updates for other V Max models as well.