Keychron/qmk_firmware #385: [Feature Request] Source code missing for several HE keyboards

Mikolaj8844, 2025-05-17:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hello,
I'm going to buy one of Keychron HE keyboards and noticed that source code for the following keyboards is missing:
- Q4 HE
- Q2 HE
- K4 HE
- K8 HE

PeterMortensen, 2025-05-27:
The same for:

 * [K4 HE](https://www.keychron.com/products/keychron-k4-he-wireless-magnetic-switch-custom-keyboard) (expected to appear *[here](https://github.com/Keychron/qmk_firmware/tree/hall_effect_playground/keyboards/keychron/k4_he)*)

PeterMortensen, 2025-06-11:
The same for:

 * [The ISO variant](https://www.keychron.com/products/keychron-q5-he-qmk-wireless-custom-keyboard-iso-layout-collection) of the Q5 HE. It isn't [here](https://github.com/Keychron/qmk_firmware/tree/hall_effect_playground/keyboards/keychron/q5_he) (as of 2025-06-11)