Keychron/qmk_firmware #386: [Feature Request] K10 QMK v2 Source Code

pdewit, 2025-05-20:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hi, the source code for the K10 QMK v2 doesn't seem to be included in the repository. Could it be added? Thank you!

PeterMortensen, 2025-05-27:
Related:

 * *[Question about the current software state #217](https://github.com/Keychron/qmk_firmware/issues/217#issuecomment-2767183831)*

PeterMortensen, 2025-06-02:
OK, they are revealed in [the JSON files](https://www.keychron.com/pages/firmware-and-json-files-of-the-keychron-qmk-k-pro-and-k-max-series-keyboards) (near *"K10 version 2 ISO RGB keymap"*) - published 10 months ago.

E.g., 

    "vendorId": "0x3434",
    "productId": "0x0DA1",

PeterMortensen, 2025-06-02:
Re *"I'm waiting for it to make the purchase"*: That was *definitely* also my intention, but my local reseller ended up only having one left in stock, and I didn't know if they would take home another batch.

apedley, 2025-06-02:
I’ve also thought about reverse engineering it. How are you planning to get the GPIO info? I didn’t see it in any of the K10 files.

PeterMortensen, 2025-06-08:
The source code for K10 V2 has [now been released](https://github.com/Keychron/qmk_firmware/tree/wls_2025q1/keyboards/keychron/k10_version_2) (both for [ISO](https://en.wikipedia.org/wiki/Keyboard_layout#Physical_layouts) and [ANSI](https://en.wikipedia.org/wiki/British_and_American_keyboards#Windows_keyboards) variants)!

PeterMortensen, 2025-06-09:
For GitHub, 2025-06-09:

To:

  <https://github.com/Keychron/qmk_firmware/issues/386>

=============================================================================

I managed to compile the K10 V2 source code and flash the keyboard with the result... And keyboard seems to work fine (this was partly typed using it).

Gotcha: 'dfu-util' does not work, because the microcontroller is , not STM. And the equivalent, 'wb32-dfu-updater_cli', must be compiled from source (on Linux).

### Procedure

 1. Installation of 'wb32-dfu-updater' (required for flashing):

        # Install prerequisites
        sudo apt-get install cmake
        sudo apt-get install libusb-1.0-0-dev # 'dev' for C development (e.g.,
                                              # header files), for the compilation
                                              # from source code (not for a
                                              # pre-alpha version...)

        # Clone
        cd $HOME
        git clone https://github.com/WestberryTech/wb32-dfu-updater.git

        # Build
        cd $HOME/wb32-dfu-updater
        sudo bash ./bootstrap.sh install

  2. Fresh install of QMK (Keychron's fork):

        cd $HOME
        qmk setup -H $HOME/Keychron_fork -b wls_2025q1 Keychron/qmk_firmware

  3. Compile (for a particular variant of the K10 V2, and with Via support):

        cd $HOME/Keychron_fork
        qmk compile -kb keychron/k10_version_2/iso/rgb -km via

        # Result
        #    72 -rwxr-xr-x 72960 Jun  9 02:21 keychron_k10_version_2_iso_rgb_via.bin

  4. Flash (for a particular variant of the K10 V2):

        cd $HOME/Keychron_fork
        wb32-dfu-updater_cli -l # Verify bootloader mode. Output should
                                # include "Found DFU: [0x342D:0xDFA0]"
        wb32-dfu-updater_cli -t -s 0x08000000 -D keychron_k10_version_2_iso_rgb_via.bin

        # Verify K10 V2 USB-side version number (set in file 'keyboard.json'
        # (<keychron/k10_version_2/iso/rgb/keyboard.json>),
        # field "device_version". Note: For a particular variant of the K10 V2
        #
        lsusb -v -d3434:0DA1 2>/dev/null | grep bcdDevice

The outout from flashing includes:

    Flash size: 256 KBytes
    SRAM size:   36 KBytes

    Download block size: 72960 Bytes

### Less space for Via macros

36 KB is significantly less than the 64 KB RAM for most other Keychron keyboard models, leaving less space for Via macros (less than 36 KB / 2 = 18 KB). The default is 1.5 KB ([exactly 2 KB]() minus an overhead of about 500 bytes)

That is presuming it uses emulated [EEPROM](https://en.wikipedia.org/wiki/EEPROM) (in flash memory). That remains to be seen. There is this in file *[info.json](https://github.com/Keychron/qmk_firmware/blob/wls_2025q1/keyboards/keychron/k10_version_2/info.json)*:

    "eeprom": {
        "driver": "i2c"
    },

Which could suggest it uses real EEPROM memory(?).

PeterMortensen, 2025-06-09:
I managed to compile the K10 V2 source code and flash the keyboard with the result... And keyboard seems to work fine (this was partly typed using it).

Gotcha: 'dfu-util' does not work, because the microcontroller is , not STM. And the equivalent, 'wb32-dfu-updater_cli', must be compiled from source (on Linux).

### Procedure

 1. Installation of 'wb32-dfu-updater' (required for flashing):

        # Install prerequisites
        sudo apt-get install cmake
        sudo apt-get install libusb-1.0-0-dev # 'dev' for C development (e.g.,
                                              # header files), for the compilation
                                              # from source code (not for a
                                              # pre-alpha version...)

        # Clone
        cd $HOME
        git clone https://github.com/WestberryTech/wb32-dfu-updater.git

        # Build
        cd $HOME/wb32-dfu-updater
        sudo bash ./bootstrap.sh install

  2. Fresh install of QMK (Keychron's fork):

        cd $HOME
        qmk setup -H $HOME/Keychron_fork -b wls_2025q1 Keychron/qmk_firmware

  3. Compile (for a particular variant of the K10 V2, and with Via support):

        cd $HOME/Keychron_fork
        qmk compile -kb keychron/k10_version_2/iso/rgb -km via

        \# Result
        \#    72 -rwxr-xr-x 72960 Jun  9 02:21 keychron_k10_version_2_iso_rgb_via.bin

  4. Flash (for a particular variant of the K10 V2):

        cd $HOME/Keychron_fork
        wb32-dfu-updater_cli -l # Verify bootloader mode. Output should
                                # include "Found DFU: [0x342D:0xDFA0]"
        wb32-dfu-updater_cli -t -s 0x08000000 -D keychron_k10_version_2_iso_rgb_via.bin

        \# Verify K10 V2 USB-side version number (set in file 'keyboard.json'
        \# (<keychron/k10_version_2/iso/rgb/keyboard.json>),
        \# field "device_version". Note: For a particular variant of the K10 V2
        \#
        lsusb -v -d3434:0DA1 2>/dev/null | grep bcdDevice

The outout from flashing includes:

    Flash size: 256 KBytes
    SRAM size:   36 KBytes

    Download block size: 72960 Bytes

### Less space for Via macros

36 KB is significantly less than the 64 KB RAM for most other Keychron keyboard models, leaving less space for Via macros (less than 36 KB / 2 = 18 KB). The default is 1.5 KB ([exactly 2 KB]() minus an overhead of about 500 bytes)

That is presuming it uses emulated [EEPROM](https://en.wikipedia.org/wiki/EEPROM) (in flash memory). That remains to be seen. There is this in file *[info.json](https://github.com/Keychron/qmk_firmware/blob/wls_2025q1/keyboards/keychron/k10_version_2/info.json)*:

    "eeprom": {
        "driver": "i2c"
    },

Which could suggest it uses real EEPROM memory(?).

PeterMortensen, 2025-06-09:
I managed to compile the K10 V2 source code and flash the keyboard with the result... And keyboard seems to work fine (this was partly typed using it).

Gotcha: '[dfu-util](https://man.archlinux.org/man/dfu-util.1.en)' does not work, because the microcontroller is *[WB32F3G71](https://github.com/Keychron/qmk_firmware/blob/wls_2025q1/keyboards/keychron/k10_version_2/info.json#L6)*, not STM. And the equivalent to dfu-util, 'wb32-dfu-updater_cli', must be compiled from source (on Linux).

### Procedure

 1. Installation of 'wb32-dfu-updater' (required for flashing):

        # Install prerequisites
        sudo apt-get install cmake
        sudo apt-get install libusb-1.0-0-dev # 'dev' for C development (e.g.,
                                              # header files), for the compilation
                                              # from source code (not for a
                                              # pre-alpha version...)

        # Clone
        cd $HOME
        git clone https://github.com/WestberryTech/wb32-dfu-updater.git

        # Build
        cd $HOME/wb32-dfu-updater
        sudo bash ./bootstrap.sh install

  2. Fresh install of QMK (Keychron's fork):

         cd $HOME
         qmk setup -H $HOME/Keychron_fork -b wls_2025q1 Keychron/qmk_firmware

  3. Compile (for a particular variant of the K10 V2, and with Via support):

         cd $HOME/Keychron_fork
         qmk compile -kb keychron/k10_version_2/iso/rgb -km via

         # Result
         #    72 -rwxr-xr-x 72960 Jun  9 02:21 keychron_k10_version_2_iso_rgb_via.bin

  4. Flash (for a particular variant of the K10 V2):

         cd $HOME/Keychron_fork
         wb32-dfu-updater_cli -l # Verify bootloader mode. Output should
                                 # include "Found DFU: [0x342D:0xDFA0]"
         wb32-dfu-updater_cli -t -s 0x08000000 -D keychron_k10_version_2_iso_rgb_via.bin

         # Verify K10 V2 USB-side version number (set in file 'keyboard.json'
         # (<keychron/k10_version_2/iso/rgb/keyboard.json>),
         # field "device_version". Note: For a particular variant of the K10 V2
         #
         lsusb -v -d3434:0DA1 2>/dev/null | grep bcdDevice

The outout from flashing includes:

    Flash size: 256 KBytes
    SRAM size:   36 KBytes

    Download block size: 72960 Bytes

### Less space for Via macros

36 KB is significantly less than the 64 KB RAM for most other Keychron keyboard models, leaving less space for Via macros (less than 36 KB / 2 = 18 KB). The default is 1.5 KB ([exactly 2 KB]() minus an overhead of about 500 bytes)

That is presuming it uses emulated [EEPROM](https://en.wikipedia.org/wiki/EEPROM) (in flash memory). That remains to be seen. There is this in file *[info.json](https://github.com/Keychron/qmk_firmware/blob/wls_2025q1/keyboards/keychron/k10_version_2/info.json)*:

    "eeprom": {
        "driver": "i2c"
    },

Which could suggest it uses real EEPROM memory(?).

PeterMortensen, 2025-06-09:
Yeah, it doesn't look promising.

From *[EEPROM Driver Configuration](https://docs.qmk.fm/drivers/eeprom#eeprom-driver-configuration)*:

> EEPROM_DRIVER = i2c. Supports writing to I2C-based 24xx EEPROM chips. See the driver section below.

And in file *[config.h](https://github.com/Keychron/qmk_firmware/tree/wls_2025q1/keyboards/keychron/k10_version_2/config.h#L25)*:

    #define EXTERNAL_EEPROM_BYTE_COUNT 2560

That is exactly 2.5 KB. Which means the overhead is 1 KB, not 0.5 KB. With dynamic per-key RGB (also stored in EEPROM memory), the space for Via macros will be reduced even further.

PeterMortensen, 2025-06-13:
I have now added most of my custom QMK things and haven't encountered any problems.

Bluetooth also works without a hitch, though [not on older versions of Linux](https://www.reddit.com/r/Keychron/comments/1g1698j/comment/m14b9bc/) (but that was expected).

PeterMortensen, 2025-06-20:
Re *"it doesn't look promising* (space for Via macros): [Perhaps it is possible](https://www.reddit.com/r/macro_pads/comments/1ge7cbi/comment/myu07jb/) to change from the physical EEPROM to emulated EEPROM, like for the other QMK-based/STM-based Keychron keyboards

The first question would be: Is that physically possible with that microcontroller?

PeterMortensen, 2025-07-31:
OK, I tried to change to emulated EEPROM in file 'info.json':

From

    "eeprom": {
        "driver": "i2c"
    },

to

    "eeprom": {
        "wear_leveling": {
            "driver": "embedded_flash",
            "logical_size": 8192,
            "backing_size": 16384
        }
    },

This resulted in a linker error, *""cannot move location counter backwards"*:

    Linking: .build/keychron_k10_version_2_iso_rgb_keychron_standardMacros.elf                          [ERRORS]
     |
     | /usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld:rules_memory.ld:314 cannot move location counter backwards (from 20007ed0 to 20007000)
     | collect2: error: ld returned 1 exit status
     |
    gmake: *** [builddefs/common_rules.mk:269: .build/keychron_k10_version_2_iso_rgb_keychron_standardMacros.elf] Error 1

Reducing it to 4.25 KB (logical size) made it compile:

    "eeprom": {
        "wear_leveling": {
            "driver": "embedded_flash",
            "logical_size": 4352,
            "backing_size": 8704
        }
    },

And Via reported *"0.0 / 2.8 KB SPACE USED"*. I haven't yet tested if all of the 2.8 KB space actually works for storing macros.

I haven't yet tested where the actual limit is between 4.25 KB and 8 KB (logical size).

### Conclusion

It seems to be possible to almost triple the space for Via macros, but 2.8 KB may still not be enough for any serious use of macros.

Synochy, 2026-10-08:
Do you think it could be possible to define another valuer for the backlighting to go off because of the battery level?
My keyboard switch off the backlight when the battery is around 40% and I think it could more like 10% before switching it off.