Keychron/qmk_firmware #393: K10 Pro coming to master- question

LucasSymons, 2025-06-18:
### Issue Description

Hi I was just wondering if k10 pro is coming to master any time soon, or going up to the upstream qmk repo?

PeterMortensen, 2025-06-20:
Coming from where exactly?

PeterMortensen, 2025-06-23:
Re *"my lighting always ended up weird"*: What kind of lighting?

It may not behave as claimed in the QMK documentation. It could be a problem with the initialisation order.

In that case, a workaround is to defer the initialisation to a few seconds after keyboard start (using a custom [EEPROM](https://en.wikipedia.org/wiki/EEPROM) (emulated in flash memory for this particular keyboard) variable to control only doing it once after a reset to factory defaults). This is to be sure the code runs last and will not be overridden by some QMK or Keychron code.

I have given up trying to make it behave as expected. I use the workaround.

closed 2025-10-22 by github-actions[bot]