Keychron/qmk_firmware #396: [Bug] Compile error "eeconfig_language.h: No such file or directory"

Crimix, 2025-07-09:
### Describe the Bug

I have been trying to compile a new version of my custom firmware for my Q5 Pro on top of the wireless playground branch. But now it fails with 

```
Compiling: keyboards/keychron/common/debounce/none.c                                                [OK]
In file included from keyboards/keychron/common/debounce/keychron_debounce.c:22:
./keyboards/keychron/common/eeconfig_kb.h:19:10: fatal error: eeconfig_language.h: No such file or directory
 #include "eeconfig_language.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling: keyboards/keychron/common/debounce/keychron_debounce.c                                   [ERRORS]
```

### Keyboard Used

Q5 Pro

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

PeterMortensen, 2025-07-11:
What source code version? [81F158](https://github.com/Keychron/qmk_firmware/commit/81f1584faab815fab228c27615391295168d65e6) (2025-06-09)?

PeterMortensen, 2025-07-11:
Related: [The source code release](https://github.com/Keychron/qmk_firmware/issues/217#issuecomment-2767183831) for the [the early 2025 Keychron keyboard main firmware updates](https://www.reddit.com/r/Keychron/comments/1j092s2/comment/ml0wljn/)

PeterMortensen, 2025-07-12:
The same happens for a K Pro series keyboard ('keychron_standardMacros' is a modified version of 'via'):

    qmk compile -kb keychron/k10_pro/iso/rgb -km keychron_standardMacros

Part of the output:

    Compiling: keyboards/keychron/common/debounce/keychron_debounce.c                                  In file included from keyboards/keychron/common/debounce/keychron_debounce.c:22:
    ./keyboards/keychron/common/eeconfig_kb.h:19:10: fatal error: eeconfig_language.h: No such file or directory
       19 | #include "eeconfig_language.h"
          |          ^~~~~~~~~~~~~~~~~~~~~
    compilation terminated.
     [ERRORS]
     |
     |
     |
    make: *** [builddefs/common_rules.mk:373: .build/obj_keychron_k10_pro_iso_rgb_keychron_standardMacros/common/debounce/keychron_debounce.o] Error 1

Thibaulltt, 2025-07-29:
I can confirm the `wls_2025q1` branch works as expected. It even allows to use the external userspace with it, which was sorely needed!

It seems like they periodically do a branch off of the main QMK repo and delete all additional keyboards in the `keyboards/` directory to release new stuff. This is ***weird***, as it does not allow to cleanly upgrade to new QMK versions, or even to track their own releases...